EdgeTX/edgetx #4773: Companion cannot download multiprotocol updates

3djc, 2024-03-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

When trying to update components, it fails on multiprotocol

### Expected Behavior

You can update components

### Steps To Reproduce

Set your update settings to include multiprotocol
Try to update components.
You will be greeted by:

[image]

### Version

Other (Please specify below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### Anything else?

Reproduced on 2.10RC1 and 2.9.4

3djc, 2024-03-19:
Totally unrelated to the issue itself, but very related to Pascal, if someone is aware of "Jarvis" sound pack, please let me or Pascal know, he is looking for one for his Iron Man theme TX16S Max.

elecpower, 2024-03-19:
Caused by the number of assets exceeding the results per page. The latest release has almost doubled the number of assets.

3djc, 2024-03-19:
Makes sense. I guess as we don't support MT12 yet on companion, we choose air stuff

closed 2024-03-24 by pfeerick

teradactyal, 2024-05-15:
Got the same problem trying to update 2.9.4 to 2.10

elecpower, 2024-05-15:
You need to update Companion to 2.10 first then update multi protocols from Companion 2.10

teradactyal, 2024-05-15:
> You need to update Companion to 2.10 first then update multi protocols from Companion 2.10

Thanks mate

PR #4774 (2024-03-24): fix(cpn): multi protocol and sd card updates
Fixes #4773

Summary of changes:
- increase the maximum results per page from 50 to 100 for multi protocol update
- use different gh media types for repo content and release asset json files (use case sd card fallback to content when no release asset found)