EdgeTX/edgetx #4786: LUA: "playnumber" no longer plays UNIT 14 (milliampere hours)

ParkerEde, 2024-03-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In the simulator, as well as in the radio firmware, the unit 14 "milliamperehours" is no longer played for a LUA widget with the command "playnumber". However, 13 percent and 15 watts work correctly.

### Expected Behavior

14 Milliamp Hour should be played

### Steps To Reproduce

try to play 14 Milliamp Hour in LUA widget

### Version

2.9.4

### Transmitter

FrSky X10 / X10S (ACCST), RadioMaster TX16S / TX16SMK2

### Anything else?

current main

philmoz, 2024-03-21:
Can you play the sound file from SD CARD file browser on the radio?

philmoz, 2024-03-21:
Also do you have both mamph0.wav and mamph1.wav on the SD card.

ParkerEde, 2024-03-21:
Sorry. I was not aware that this unit is now differentiated between single and multiple numbers. it works with mamph1.wav. Thank you

closed 2024-03-21 by ParkerEde