EdgeTX/edgetx #4789: More brightness settings for OLED screen

KondratokGK, 2024-03-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I want for full brightness on OLED when i use it. But OLED screen wear out on high brightness.

### Describe the solution you'd like

It should change low brightness value, inactivity time from to low brightness and condition for full brightness(All/buttons/trims/alarm etc.). Just copy same menu from monochrome lcd and add some modifications.

### Describe alternatives you've considered

1. Use knob for brightness

### Additional context

Now i use jumper t20 gemini v1 or v2 IDK and v2.10.0-rc1.

3djc, 2024-03-22:
So what exactly are you missing??? What you described is already part of t20 firmware since the beginning

3djc, 2024-03-22:
Ok, I see what you mean, the feature is there, but the settings have disappeared, will make sure they are back before next RC

PR #4793 (2024-03-24): feat: allow configuration of auto dimming for OLED screen
Allow configuration of the Oled equivalent of LCD backlight control, where Oled will dim to the minimum where LCD equivalent backlight off.

Fixes #4789
closed 2024-03-24 by pfeerick