EdgeTX/edgetx #4792: stdbool.h missing when compiling on KDE Neon 6.0

martonmiklos, 2024-03-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

Building the current master fails with:

```
[ 35%] Generating datacopy.inc
cd /home/mm/Projektek/edgetx/build-output/arm-none-eabi/radio/src && /usr/bin/python3 /home/mm/Projektek/edgetx/radio/util/generate_datacopy.py /home/mm/Projektek/edgetx/radio/src/datastructs_private.h -DBACKUP -D_GLIBCXX_USE_C99=1 -DPCBFRSKY -DSOFTWARE_VOLUME -DRADIO_TX16S -DRADIO_FAMILY_T16 -DMANUFACTURER_RADIOMASTER -DPCBX10 -DPCBREV=TX16S -DPCBREV_TX16S -DPCBHORUS -DSTM32F429_439xx -DSTM32F429xx -DSDRAM -DCCMRAM -DCOLORLCD -DHARDWARE_KEYS -DEEPROM_VARIANT=0 -DAUDIO -DVOICE -DRTCLOCK -DGPS_USART_BAUDRATE=9600 -DPWR_BUTTON_PRESS -DSTM32_SUPPORT_32BIT_TIMERS -DDEFAULT_THEME_DARKBLUE -DCONFIGURABLE_MODULE_PORT -DACCESS_DENIED -DDISK_CACHE -DINTERNAL_GPS -DHARDWARE_TOUCH -DSWSERIALPOWER -DUSB_CHARGER -DMULTI_PROTOLIST -DTHREADSAFE_MALLOC -DMAX_FLEX_SWITCHES=2 -DSTM32F4 -DSTM32 -DLUA_INPUTS -DVARIO -DIMU -DAUX_SERIAL -DAUX2_SERIAL -DFLYSKY_GIMBAL -DUSB_SERIAL -DHARDWARE_INTERNAL_MODULE -DINTERNAL_MODULE_PXX1 -DHARDWARE_INTERNAL_MODULE -DINTERNAL_MODULE_PXX2 -DHARDWARE_INTERNAL_MODULE -DINTERNAL_MODULE_MULTI -DINTERNAL_MODULE_SERIAL_TELEMETRY -DHARDWARE_INTERNAL_MODULE -DINTERNAL_MODULE_CRSF -DINTERNAL_MODULE_SERIAL_TELEMETRY -DDEFAULT_INTERNAL_MODULE=MODULE_TYPE_MULTIMODULE -DHARDWARE_EXTERNAL_MODULE_SIZE_STD -DHARDWARE_EXTERNAL_MODULE -DTIMERS=3 -DCLI -DENABLE_SERIAL_PASSTHROUGH -DHAPTIC -DPPM -DDSM2 -DSBUS -DPXX -DPXX1 -DPXX -DPXX2 -DMULTIMODULE -DAFHDS3 -DCROSSFIRE -DGHOST -DCPUARM -DGPS -DBATTGRAPH -DTHRTRACE -DTRANSLATIONS_EN -DTRANSLATIONS="EN" -DTRANSLATION_IS_RTL=0 -DFLAVOUR="tx16s" -DLIBOPENUI -DSDCARD_YAML -DSTORAGE_MODELSLIST -DSTORAGE_MODELSLIST -DRTC_BACKUP_RAM -I/home/mm/Projektek/edgetx/radio/src/thirdparty/FatFs -I/home/mm/Projektek/edgetx/radio/src/thirdparty/FatFs/option -I/home/mm/Projektek/edgetx/radio/src/thirdparty/STM32F4xx_HAL_Driver/Inc -I/home/mm/Projektek/edgetx/radio/src/thirdparty/STM32F4xx_DSP_StdPeriph_Lib_V1.8.0/Libraries/STM32F4xx_StdPeriph_Driver/inc -I/home/mm/Projektek/edgetx/radio/src/thirdparty/STM32F4xx_DSP_StdPeriph_Lib_V1.8.0/Libraries/CMSIS/Device/ST/STM32F4xx/Include -I/home/mm/Projektek/edgetx/radio/src/thirdparty/STM32F4xx_DSP_StdPeriph_Lib_V1.8.0/Libraries/CMSIS/Include -I/home/mm/Projektek/edgetx/radio/src/thirdparty/FreeRTOS/portable/GCC/ARM_CM4F -I/home/mm/Projektek/edgetx/radio/src/targets/common/arm/stm32 -I/home/mm/Projektek/edgetx/radio/src/targets/horus -I/home/mm/Projektek/edgetx/radio/src/thirdparty -I/home/mm/Projektek/edgetx/radio/src -I/home/mm/Projektek/edgetx/build-output/arm-none-eabi/radio/src -I/home/mm/Projektek/edgetx/build-output/arm-none-eabi/radio/src/bitmaps/480x272 -I/home/mm/Projektek/edgetx/radio/src/gui/colorlcd -I/home/mm/Projektek/edgetx/radio/src/gui/colorlcd/layouts -I/home/mm/Projektek/edgetx/build-output/arm-none-eabi/radio/src/fonts -I/home/mm/Projektek/edgetx/radio/src/thirdparty/libopenui/src -I/home/mm/Projektek/edgetx/radio/src/thirdparty/libopenui/thirdparty -I/home/mm/Projektek/edgetx/radio/src/gui -I/home/mm/Projektek/edgetx/radio/src/gui/colorlcd -I/home/mm/Projektek/edgetx/radio/src > datacopy.inc
/home/mm/Projektek/edgetx/radio/src/definitions.h:29:10: fatal error: 'stddef.h' file not found
  /home/mm/Projektek/edgetx/radio/src/datastructs_private.h:26:10: note: in file included from /home/mm/Projektek/edgetx/radio/src/datastructs_private.h:26:
  /home/mm/Projektek/edgetx/radio/src/targets/horus/board.h:25:10: note: in file included from /home/mm/Projektek/edgetx/radio/src/targets/horus/board.h:25:
make[7]: *** [radio/src/CMakeFiles/firmware.dir/build.make:87: radio/src/datacopy.inc] Error 255
```
I have installed everything with the setup_buildenv_ubuntu22.04.sh 

### Expected Behavior

Code got compiled ;)

### Steps To Reproduce

Install KDE Neon 6.0
Run setup_buildenv_ubuntu22.04.sh 
 cmake -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Debug ../
make firmware VERBOSE=1

### Version

2.9.4

### Transmitter

BetaFPV LiteRadio3 Pro

### Operating System (OS)

Linux

### OS Version

KDE neon

pfeerick, 2024-03-23:
FYI, the only really supported Linux distrbution is Ubuntu, v20.04 IIRC.
Even on that distribution, I use a docker container to ensure everything is
matched properly and doesn't break, and obviously even more helpful if not
running that distro. If you install vscode, you can use the devcontainer
functionality to get this running (just need to install docker as
prompted). Or run it directly and use code editor of your choice. Another
possibility is to use gitpod.

Check the wiki for some of the guides -
https://github.com/EdgeTX/edgetx/wiki

3djc, 2024-03-23:
Precision: Ubuntu 22.04 for 2.10 and main branch

pfeerick, 2024-03-24:
Ok, I can see that KDE neon 6 is based on ubuntu 22.04, meaning it should be (and is) a usable base OS. I've just installed it in a VM, and have successfully built a firmware for the TX16s, so I think you must have missed a step somewhere. 

i.e. if you didn't run `source ~/.bashrc"` as prompted at the very end of the run of `setup_buildenv_ubuntu22.04.sh`, the compiler toolchain wouldn't have been in your path. Simply closing and re-opening the terminal would also clear that unless you're using a different shell. I'm not sure you would have gotten even that far if that was missing though. ls

Since you said 2.9.4... did you check out the `2.9` branch, or `main`? `main` should work still, but you would have gotten a warning that PPM_UNIT is not used by the project. 

Also, did you run either `make configure` or `make arm-none-eabi-configure` before `make firmware`? 

Since you used the example cmake for the TX16S, but listed the LR3Pro, I'm going to assume that mismatch is because you're trying to get it working still. But intend to build for the LR3Pro?

pfeerick, 2024-03-24:
```
pfeerick@vbox-kdeneon6:~/repos/edgetx/edgetx/build$ lsb_release -a
No LSB modules are available.
Distributor ID: Neon
Description:    KDE neon 6.0
Release:        22.04
Codename:       jammy
pfeerick@vbox-kdeneon6:~/repos/edgetx/edgetx/build$ cmake -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Debug ../
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_ARGS: -DDEFAULT_MODE=2;-DGVARS=YES;-DLUA_MIXER=YES;-DPCB=X10;-DPCBREV=TX16S;-DPPM_UNIT=US
-- CMAKE_BUILD_TYPE: Debug
-- Configuring done (0.3s)
-- Generating done (0.0s)
-- Build files have been written to: /home/pfeerick/repos/edgetx/edgetx/build
```
```
pfeerick@vbox-kdeneon6:~/repos/edgetx/edgetx/build$ make arm-none-eabi-configure[ 20%] Creating directories for 'arm-none-eabi'
[ 40%] No download step for 'arm-none-eabi'
[ 60%] No update step for 'arm-none-eabi'
[ 80%] No patch step for 'arm-none-eabi'
[100%] Performing configure step for 'arm-none-eabi'
loading initial cache file /home/pfeerick/repos/edgetx/edgetx/build/arm-none-eabi-prefix/tmp/arm-none-eabi-cache-Debug.cmake
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3") 
-- Python found, version: 3.10.12
-- EdgeTX 2.9.4-selfbuild @ f0a50bd9
-- Internal GPS enabled
-- Adding support for USB serial
-- Adding support for PXX1 as internal module
-- Adding support for PXX2 as internal module
-- Adding support for MULTI as internal module
-- Adding support for CRSF as internal module
Target has SDRAM, do not use BIN_ALLOCATOR
TARGET bootloader: cpp compiler /opt/gcc-arm-none-eabi/bin/arm-none-eabi-g++ v10.2.1
TARGET firmware: cpp compiler /opt/gcc-arm-none-eabi/bin/arm-none-eabi-g++ v10.2.1
-- Configuring done (0.3s)
-- Generating done (0.1s)
-- Build files have been written to: /home/pfeerick/repos/edgetx/edgetx/build/arm-none-eabi
[100%] Built target arm-none-eabi-configure
```

```
pfeerick@vbox-kdeneon6:~/repos/edgetx/edgetx/build$ make -j`nproc` firmware
... clipped as it's so long ... no effective difference with or without -j for me other than build time
[100%] Built target arm-none-eabi-configure
[... 94 lines cut ...]
firmware.elf  :
section               size         addr
.text              1614768    134217728
.ARM                     8    135832496
.init_array            248    135832504
.fini_array              4    135832752
.data                 3324    536870912
.bss                 50224    536874240
.ram                 20160    536924464
.ccm                 48284    268435456
.sdram             5238784   3489660928
.ARM.attributes         48            0
.debug_info       20078268            0
.debug_abbrev      1137167            0
.debug_loc         3852323            0
.debug_aranges      100200            0
.debug_ranges       672432            0
.debug_line        2863049            0
.debug_str         2802459            0
.comment                77            0
.debug_frame        400748            0
Total             38882575

[100%] Built target firmware-size
[100%] Built target firmware-size

raphaelcoeffic, 2024-03-30:
@martonmiklos the issue is most probably caused by the way we search for GCC headers to feed them to `libclang`, as it doesn't know how to locate these headers by default.

If you want the easy, just install `docker` and use our build containers.

If you're trying to get yourself a dev environment, dig into `find_clang.py`, and there have a special look at `getBuiltinHeaderPath`. This is most likely where the issue is.

martonmiklos, 2024-03-30:
Hi @pfeerick 

Many thanks for your efforts put into troubleshooting. 
I will double check if there is anything I missed.

> Since you used the example cmake for the TX16S, but listed the LR3Pro, I'm going to assume that mismatch is because you're trying to get it working still. But intend to build for the LR3Pro?

One of my friends asked me make a custom build to his remote and I wanted to do a build feasibility test first (I do not even know what is the target device yet).

martonmiklos, 2024-04-20:
> @martonmiklos the issue is most probably caused by the way we search for GCC headers to feed them to `libclang`, as it doesn't know how to locate these headers by default.
>
> If you want the easy, [...]

The hits were correct, I managed to fix it.

On KDE NEON only the following libclang.so containing packages can be found:
```
mm@P15:~/Projektek/edgetx/radio/util$ apt-file search libclang.so
libclang-11-dev: /usr/lib/llvm-11/lib/libclang.so
libclang-12-dev: /usr/lib/llvm-12/lib/libclang.so
libclang-13-dev: /usr/lib/llvm-13/lib/libclang.so
libclang-14-dev: /usr/lib/llvm-14/lib/libclang.so
libclang-15-dev: /usr/lib/llvm-15/lib/libclang.so
libclang1-11: /usr/lib/llvm-11/lib/libclang.so.1
libclang1-12: /usr/lib/llvm-12/lib/libclang.so.1
libclang1-13: /usr/lib/llvm-13/lib/libclang.so.1
libclang1-14: /usr/lib/llvm-14/lib/libclang.so.1
libclang1-15: /usr/lib/llvm-15/lib/libclang.so.1
```

find_clang.py does not find it as these paths are hardcoded:
https://github.com/EdgeTX/edgetx/blob/main/radio/util/find_clang.py#L71

For clang 14 some additional include paths needs to be added for correct datacopy build:
https://github.com/martonmiklos/edgetx/blob/fix_kde_neon_build_failure/radio/util/generate_datacopy.py#L104

I have pushed my fixes to my fork, if you feel it has added value I could submit a PR:
https://github.com/EdgeTX/edgetx/compare/main...martonmiklos:edgetx:fix_kde_neon_build_failure?expand=1

raphaelcoeffic, 2024-04-21:
@martonmiklos we should find a more generic solution.... it is really annoying to have to hunt for that library's path ....

martonmiklos, 2024-04-21:
> @martonmiklos we should find a more generic solution.... it is really annoying to have to hunt for that library's path ....

@raphaelcoeffic I am new to here, and unsure about the rules to play with. I can implement a better clang detector, but what should be the goal: selecting the minimum or the maximum available version?