EdgeTX/edgetx #4797: x9d+2019  - main view not showing/updating pot/slider bars correctly

pfeerick, 2024-03-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On the X9D+2019 (but NOT the x9dp) the upper slider bars do not update properly on the main view and are mixed up... the sliders used to be at the bottom, and the pots at the top. Maybe something to do with it only having no S3 unlike the 

From the radio, with two sliders configured:
[image]

Simulator, which also shows the wrong one is moving:
[image]

### Expected Behavior

That the pot/slider bars show/work and be in the right place! My needs are very simple :hand_over_mouth: 

I was going to say that they sould be turned on and off depending on how the controls are configured (i.e. hidden if none) but that would be new behaviour, not a bugfix, so unimportant. 

### Steps To Reproduce

Turn on the radio / run a blank/unconfigured simulator. 

### Version

Other (Please specify below)

### Transmitter

FrSky X9D+2019

### Anything else?

v2.10.0-rc1

For what it is worth, on the X9D+, also running 2.10.0-rc1, everything works fine. I did notice in SYS -> HARDWARE, it shows configuration options for a third pot, S3 (i.e. S1/S2/S3/LS/RS), whereas the X9D+2019 only shows S1/S2/LS/RS - which didn't use to be the case.

kazuzaPT, 2024-05-19:
Confirm the exact same issue on v2.10.0 stable release

Mmarsian, 2024-06-07:
I've just flashed 2.10.1 and am also confirming the issue. Right slider is now on top left and left slider is not shown.

closed 2024-06-11 by pfeerick