EdgeTX/edgetx #4798: commando8 - does not power on

pfeerick, 2024-03-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After installing the 2.10.0-rc1 bootloader/firmware, the iFlight Commando8 will not boot into the main firmware properly. The bootloader works fine. The firmware gets as far as turning the backlight on, setting the power led to blue, and then seems to cut out just as the haptic feedback would occur. 

@raphaelcoeffic I believe you had a handset also? A reminder, power control on this radio is... unique... as it has it's own MCU and startup delay/sequence. 

### Expected Behavior

That, you know, the radio actually turn on?

### Steps To Reproduce

1. Press/release the power button to get the first stage active... battery status leds will light up to show battery level.
2. Press and hold the power button while the battery status leds are active... they will go out and then progressively light up. Once all four are lit, the backlight and blue power light will come on, and immediately go out. 
3. The battery leds will most likely go out after a short time, as if the radio had shut down. 

### Version

Other (Please specify below)

### Transmitter

iFlight Commando8

### Anything else?

v2.10.0-rc1

I will try running a bisect to see where the breakage started. Since the bootloader is still functioning, I can revert to 2.9.4 and that works fine.

pfeerick, 2024-04-16:
Sorry @raphaelcoeffic ... git bisect led me to this... 

```
5153aaeea074a8427ab55493b6958b53ef88c46e is the first bad commit
commit 5153aaeea074a8427ab55493b6958b53ef88c46e
Author: Raphael Coeffic <raphaelcoeffic@users.noreply.github.com>
Date:   Sun Dec 3 22:14:25 2023 +0100

    refactor: IWDG driver and EM trigger (#3846)
 ```
 
 I noticed the Commando8 is the only target to define 
```
set(USE_RTC_CLOCK NO)
```
so tried setting that for the Pocket, but it worked just fine still, so all yours :zany_face: Let me know if there is anything further I can test. I'm running on hotspot internet atm as internet is currently completely out after two days of dialup speeds... so much for upgrades...

pfeerick, 2024-04-25:
Bump @raphaelcoeffic :-P

richardclli, 2024-05-09:
USE_RTC_CLOCK seems not related to elsewhere??!! It should change the definition RTCLOCK otherwise it is useless.

richardclli, 2024-05-09:
What MCU is commando8 using?

pfeerick, 2024-05-09:
https://github.com/EdgeTX/edgetx/blob/96a61876b79a2ec477e493782c89bace091c55cd/radio/src/targets/taranis/CMakeLists.txt#L348

Raphael has it on his homework list ;)

uberhalit, 2024-07-02:
Any news on this @pfeerick ?

raphaelcoeffic, 2024-07-03:
It seems to be some strange behaviour of the power button (controlled by a separate microcontroller).

If I use this sequence:
- short power button press,
- then long button press (until screen comes up)

It works for me. Same applies to power down. When pressing power button while the firmware is on, it seems at first the button does not work. However, if you just repeat the same sequence (short press, then long one), it works as usual.

pfeerick, 2024-07-03:
No, no change. I was about to say that something has changed between 2.10.0 and 2.10.2, as it booted up just fine after flashing 2.10.2, but after a second flash between 2.10.0 and 2.10.2, it is behaving exactly as described in the initial issue again. So going back to 2.9.4. So it looks like https://github.com/EdgeTX/edgetx/issues/4798#issuecomment-2058620169 is still relevant. 

@raphaelcoeffic That is the expected behaviour - it uses the DJI style single press to get battery level or press, release and then hold to power on (or off) while the . You also use pull the left navigation hat and hold the power button to enter the bootloader, or connect USB and do the same to enter DFU mode. As well as a double press of the power button to enable or disable USB while firmware is running and USB is plugged in.

raphaelcoeffic, 2024-07-03:
Ok, I just tried to flash back to 2.10.0 (booting normally), and back to 2.10.2, and everything is still normal 🙃

pfeerick, 2024-07-03:
Oh fun, hope this isn't good to be like the TX16S, and only resolve after
full chip erase, like 1-2 users reported. This was with DFU flashing?

raphaelcoeffic, 2024-07-04:
Yes, DFU or SWD.

pfeerick, 2024-07-04:
@uberhalit An update for you, as Raphael already knows :hand_over_mouth: ... I finally managed to get mine to stop playing silly buggers and consistently power on with 2.10.x firmware, so there will hopefully be a fix for it in 2.10.3.

PR #5245 (2024-07-04): fix: Commando8 does not need the startup animation
Fixes #4798 since startup animation pwrOff() was responsible for shutting down radio
Fixes #4995 since screen was not being cleared on power up for radios with power switches
closed 2024-07-04 by pfeerick