EdgeTX/edgetx #4799: Jumper T16 scroll wheel not debounced

gazhayes, 2024-03-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After updating from openTX 2.3.6 the scroll wheel on the Jumper T16 appears to no longer be debounced. 

This means that it can only go back or forward one item (so the menus cannot be used other than the 1st and 2nd item in whatever list is being scrolled).

### Expected Behavior

The scroll wheel should allow navigation through all items in any given menu.

### Steps To Reproduce

Just try using the latest firmware on a T16.

### Version

2.9.4

### Transmitter

Jumper T16

### Operating System (OS)

macOS

pfeerick, 2024-03-28:
Just appending the further information that was provided in the other issue:

> Ok I've flashed 2.7.1 and now instead of going back and forth between two menu items I can make it move around more by flicking the wheel faster (this was not possible with 2.9.x).
> 
> It was 100% working with 2.3.6 so it's probably not strictly a hardware issue.
> 
> On the key diagnostics page, the rotary encoder value alternates between 0 and 2 with each click of the wheel. 

Can I just clarify/confirm that "can only go back or forward one item" applies *everywhere*... i.e. main menu carousel, trying to scroll down model and radio settings menus, as well as for navigating popup menus?

gazhayes, 2024-03-28:
> Can I just clarify/confirm that "can only go back or forward one item" applies everywhere...

Yep, exactly. But I now have evidence this is actually a hardware issue, it started working perfectly for a few minutes and then reverted back to broken behaviour. 

Any hardware suggestions regarding the ribbon cable?

pfeerick, 2024-03-28:
Ouch! All I can suggest is try re-seating/swapping ends for the "ribbon cable" - flat flex cable (FFC), check the contacts for any signs of tarnish/corrosion, inspect the FFC for any damage. It looks like on the T16 it goes from the roller board to a sub-board which also does all the switches, and then there is a wide ribbon / flat flex cable that goes to the mainboard, so you will want to check both. But should be able to start with just the roller board first, as the wide one will be a pain to get at, and is *probably* fine. 

Did you ever get the replacement ribbon cables from Jumper for the T16? As they had "free" (bar shipping, I think) replacement cables to replace the substandard ones they originally used. 

https://www.youtube.com/watch?v=DwAc9lr1iBM

https://www.youtube.com/watch?v=qdf1QPlagJU

closed 2025-03-15 by github-actions[bot]