EdgeTX/edgetx #4802: MacOS 14.4 Sonoma OpenTX Companion crashes on Opening from versions 1.9.0 onwards.

BJR981S, 2024-03-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

My current version works fine with known bugs. 1.8.4

On upgrading to the version from 1.9.0 to 1.9.4 It traps on opening the application.

### Expected Behavior

Expect it not to crash when opening application.

### Steps To Reproduce

Open application after install.

### Version

2.9.4

### Transmitter

FrSky X9D+, FrSky X12, RadioMaster TX16S / TX16SMK2

### Operating System (OS)

macOS

### OS Version

Version 14.4 Sonoma.

3djc, 2024-03-24:
Working fine here

[image]

[image]

pfeerick, 2024-03-24:
I presume the 1.8.4, 1.9.0 and reference to OpenTX companion were typos?

BJR981S, 2024-03-26:
No. 1.8.4 Works normally on Mac OS 14.4 1.9.0, 1.9.1, 1.9.2, 1.9.3 and 1.9.4 all trap on opening. I tested all individually.

[image]
[image]
[image]

3djc, 2024-03-26:
I'm running Apple silicon (M2 Max), that might be why it works fine on my mac

BJR981S, 2024-03-26:
Quite possible. EdgeTX 1.9.0 onwards may be built on AppleSilicon binaries only. Would need an X86/X64 binary build, on that binaries it would still run on Apple silicon through legacy x64 support.

I will try on an earlier version of OSX. If it fails there then you assumption would be correct. I will advise once I test.

philmoz, 2024-03-26:
I have and iMac Pro and all versions of Companion run fine for me. 

It is crashing in the SDL library while initialising joysticks.

Companion 2.9.x uses SDL 2, whereas 2.8 and earlier used SDL 1.2.

If you have any joysticks connected to your Mac try unplugging them first.

pfeerick, 2024-03-26:
> No. 1.8.4 Works normally on Mac OS 14.4 1.9.0, 1.9.1, 1.9.2, 1.9.3 and 1.9.4 all trap on opening. I tested all individually.

The reason I asked is because you wrote 1.9.4, instead of 2.9.4... and have been consistently writing 1.x instead of 2.x, and then said OpenTx companion in the title ;)

BJR981S, 2024-03-27:
Ah silly me. yes that would be a brain fart.

I tested 2.9.4 on MAcOS Ventura and it installed and ran fine on an X86/X64 MacBook. So not an apple silicon issue. Must have something to do with either my machine or the upgrade to 2.9.4 from 2.8.4 from within the application. I deleted both 2.9.4 and 2.8.4 from my machine and purged the application caches and installed 2.9.4 again but it has the same result. I have started 2.9.4 from inside the application contents folder and have the following error messages available.

Last login: Tue Mar 26 18:19:42 on console
/Applications/EdgeTX\ Companion\ 2.9.app/Contents/MacOS/companion ; exit;
brian@imacpro ~ % /Applications/EdgeTX\ Companion\ 2.9.app/Contents/MacOS/companion ; exit;
[I] AppData::init():536 - Settings init with EdgeTX Companion 2.9 Saved version: 2090402 Current version: 2090402
[D] Translations::installTranslators():104 - Locale name: "en_AU" language: "Australian English" country: "Australia"
[W] Translations::tryLoadTranslator():165 - Error installing translation file "qt" for: "en_AU" from: ":/translations"
[W] Translations::tryLoadTranslator():165 - Error installing translation file "qtbase" for: "en_AU" from: ":/translations"
[D] Translations::tryLoadTranslator():161 - Installed translation file "companion" from ":/translations" for "en_AU"
companion(20462,0x7ff85dfe5100) malloc: Heap corruption detected, free list is damaged at 0x6000010b0f80
*** Incorrect guard value: 11177934275134619647
companion(20462,0x7ff85dfe5100) malloc: *** set a breakpoint in malloc_error_break to debug
zsh: abort      /Applications/EdgeTX\ Companion\ 2.9.app/Contents/MacOS/companion

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...       3 completed.

[Process completed]

BJR981S, 2024-03-28:
Ok I have now tried 3 different MacOS machines at different version levels. And all work fine except for my iMac Pro  so there is something specific about it. I was running 2.8.4 So I uninstalled it and all other versions. And reinstalled it again. Its worked fine. Out of interest I upgraded to 2.8.5 and that worked fine as well. I can only think that there is something in the data files that is preventing 1.9.0 and above from working. I even tried the new RC1 That failed as well. I have searched around for the folder where the application data is stored but I cant find it. Can someone give me the folder path on OSX to the edgeTX data files location? I want to try a completely clean installation.

BJR981S, 2024-03-28:
> I have an iMac Pro and all versions of Companion run fine for me.
> 
> It is crashing in the SDL library while initialising joysticks.
> 
> Companion 2.9.x uses SDL 2, whereas 2.8 and earlier used SDL 1.2.
> 
> If you have any joysticks connected to your Mac try unplugging them first.

I have joysticks and multiple keyboards I also have Razor keypads that act as joysticks.

I will give your suggestion a try.

Bt if that is the case that becomes a problem in using the Edge app.