EdgeTX/edgetx #4803: PT support: Missing special characters support

zandorsp, 2024-03-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

[image]

There is no support for special characters in Portuguese. The radio replace then with a space character.

### Expected Behavior

The Portuguese( PT ) needs the following special characters: 
Á, á
Ã, ã
Â, â
Ç, ç
Ê, ê
É, é
Í, í
Î, î
Ó, ó
Ô, ô
Õ, õ
Û, û
Ú, ú

### Steps To Reproduce

Try to translate the PT.h file with special characters

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX12 / TX12MK2

### Anything else?

no

philmoz, 2024-03-24:
It looks like this was started at some point;  but never completed.

To support special characters on B&W radios there needs to be image files with the special characters drawn into them.
These are in the radio/src/fonts/std and radio/src/fonts.sqt5 folders.

For PT these would be he font_pt_04x06.png, font_pt_05x07.png, font_pt_08x10.png and font_pt_10x14.png files.
The numbers are the font dimensions in pixels.

The 4x6 font file is empty. The others contain some of the characters you listed; but not all of them.

3djc, 2024-03-24:
Think they are in 
[image]

but yes, 4x6 is empty

zandorsp, 2024-03-25:
Yes the font_pt_05x07.png, font_pt_08x10.png are right and contain all needed characters

3djc, 2024-03-25:
@zandorsp could you check: https://github.com/EdgeTX/edgetx/pull/4806 ? I think it fixes your issue

PR #4806 (2024-03-27): fix(bw): missing PT special chars handling

[image]

Fixes #4803
closed 2024-03-27 by pfeerick