EdgeTX/edgetx #4805: lr3pro - allows shutdown while rx connected without waiting for confirmation

pfeerick, 2024-03-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If you try to turn off the LR3Pro while a telemetry receiver is still active, it will show the "model still powered" message during the shutdown animation phase, but rather than show the "shutdown?" confirmation prompt, it switches off once the shutdown animation timeout is reached. 

### Expected Behavior

That it show the "shutdown?" confirmation prompt rather than power down.

### Steps To Reproduce

1. With a telemetry receiver bound and active (in my case EP2), hold down the power button on the radio to turn it off.
2. Once the shutdown animation is complete, witness the radio power off instead of show the shutdown confirmation prompt.

### Version

Other (Please specify below)

### Transmitter

BetaFPV LiteRadio3 Pro

### Anything else?

This is not new behaviour - has been present for a little while as I remember it in 2.9 and is present in 2.8.4 also... so possibly ever since the lr3pro was added. 

@3djc Are there transmitter specific defines/code could be the culprit here?

philmoz, 2024-03-24:
I assume that 'RSSI shutdown' option is on.

[image]

pfeerick, 2024-03-25:
Good _not_ to assume... looks like it turned itself off when the settings were getting corrupted before. It was the discussion about that setting the other day that brought this to mind again but I forgot to check before posting. Thanks!

closed 2024-03-25 by pfeerick