EdgeTX/edgetx #4807: Zorro Telemetry Mirror not working on 2.10RC1

Pelado-Mat, 2024-03-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Since updating to 2.10RC1 Telemetry mirror on AUX1 is not working. 
It's working fine on 2.9.X

### Expected Behavior

Telemetry mirror to work

### Steps To Reproduce

1. On 2.9.X Check that Telemery Mirror is working
2. Update to 2.10.0RC1
3. Check if Telem Mirror works
    No Telemetry is mirrored to the port

### Version

Other (Please specify below)

### Transmitter

RadioMaster Zorro

### Operating System (OS)

Linux

Pelado-Mat, 2024-03-31:
The same happens on RC2

pfeerick, 2024-03-31:
Can I get a few more details / specifics please? MPM (and what protocol) or ELRS? What are you using to monitor the AUX1 - is this just a USB UART, bluetooth adapter, something else?

Pelado-Mat, 2024-03-31:
> Can I get a few more details / specifics please? 
>MPM (and what protocol) or ELRS? 

ELRS

> What are you using to monitor the AUX1 - is this just a USB UART, bluetooth adapter, something else?

A BT Adapter HC05 - JDY-31-v1.35
BaudRate 57600

The details of the configuration are [here](https://aricafpv.cl/es/Biblioteca/ZorroMods#bt-mod)

For reference, if I downgrade to EdgeTX 2.9.X the telem mirror works fine.

As the app to view the telemetry I'm using [RomanLut's app](https://github.com/RomanLut/android-taranis-smartport-telemetry)

Let me know how I can help testing.

3djc, 2024-04-03:
Which explains the issue is still opened...

3djc, 2024-04-09:
ok, trying to reproduce on Boxer, working fine there on main. Trying on Zorro

3djc, 2024-04-09:
I'm sorry to report that it works fine here

[image]

[image]

Aux1 set to telemetry mirror, on a CRSF link, as far as I can tell, it works just perfectly fine.

Thanks to Radiomaster for sending me a production Zorro so I could test this

Pelado-Mat, 2024-04-10:
I still have the same behavior. All works fine on 2.9.X, if I upgrade to 2.10RC the app does connect but  does notget any telemetry.
I open an issue on the app github RomanLut/android-taranis-smartport-telemetry#53

Pelado-Mat, 2024-04-10:
Im trying this to capture the serial stream i got via BT:
1. Flash the radio with 2.9.4
1. Start the radio and use a known working model
1. Connect my linux PC via BT to the radio
1. Start capturing the serial connection:
` stdbuf -o0 cat /dev/rfcomm0 > tmp/2.9.4.log `
1. Plug the model to get some telemetry
1. Leave the capture some time
1. stop the capture, disconnect the PC
1. Connect the android app to double check that is working OK or not

I repeat the same procedure with 2.10rc

here are both files that I captured
[2.10.log](https://github.com/RomanLut/android-taranis-smartport-telemetry/files/14934687/2.10.log)
[2.9.4.log](https://github.com/RomanLut/android-taranis-smartport-telemetry/files/14934685/2.9.4.log)

Pelado-Mat, 2024-04-10:
After some troubleshooting with @3djc help,  we find that the BT module baud rate was the issue here.
On 57600  works fine up to 2.9.4 but not in 2.10rc. 
On 115200 works fine on 2.10rc but not in 2.9 or earlier.

Roman-Golod, 2024-04-10:
So what should we do, flash out bluetooth modules or wait for fix?

Pelado-Mat, 2024-04-10:
> So what should we do, flash out bluetooth modules or wait for fix?

Could you reproduce the issue? As far as I understand @3djc could not.

pfeerick, 2024-04-11:
Next step is probably to figure out why the baud rate has changed.
Bluetooth module should be arriving today so should be able to test this
using same setup as you soon.

Roman-Golod, 2024-04-11:
> > So what should we do, flash out bluetooth modules or wait for fix?
> 
> Could you reproduce the issue? As far as I understand @3djc could not.

Yes, I can confirm, that the telemetry mirroring works in 2.9.4 and doesn't work in 2.10 rc3.

3djc, 2024-04-11:
It should have been 115k anyway from day 1 anyway. Will do some more test later today, but 115k is what it needs to be. Why some radio do not have it in 2.9 I'm not to sure yet

3djc, 2024-04-11:
Was able to reproduce (seems to be bootloader related). Anyhow, it would be in the ‘fix’ category, since it was abnormal to not use 115k

Roman-Golod, 2024-04-11:
> Was able to reproduce (seems to be bootloader related). Anyhow, it would be in the ‘fix’ category, since it was abnormal to not use 115k

Agree with you about 115k. Last yeaar some people added BT in their TXes and everywhere it was 115k except Zorro and TX12mk2.

3djc, 2024-04-11:
@pfeerick imho, this should be closed as it is normal behavior, and maybe note should be added in 2.10 release note that CRSF telemetry mirror has been aligned with other radios to 115k for Zorro and TX12MK2

pfeerick, 2024-04-11:
_Just_ CRSF, or all telemetry mirror?

3djc, 2024-04-11:
Just CRSF, others (namely frsky) use 57k because of sport and frsky software

Thidsa, 2024-04-18:
Here on TX16s with 2.9.4, telemetry mirror aux1 dont work with ext. tbs crossfire module. With a ext Radiomaster Ranger it works.
Same hc05 as @Pelado-Mat. Only know it worked with both before. Cant remember when, sometimes last year.

dcan999, 2024-07-22:
> I'm sorry to report that it works fine here [...]

Sorry for the off topic question, but where do you connect on the Zorro for Aux 1?

Pelado-Mat, 2024-07-22:
There are two pads on the back of the mainboard.

https://www.aricarc.cl/es/Biblioteca/ZorroMods#bt-mod

dcan999, 2024-07-24:
> There are two pads on the back of the mainboard. https://www.aricarc.cl/es/Biblioteca/ZorroMods#bt-mod
> […](#)

Thank you. I tried this today, but my HC-05 will not work powered by 3.3v for some reason. When I power it by external 5v it works fine. Is there any 5v power source in the zorro?

dcan999, 2024-07-24:
Upon further testing, (with 5v), there is no telemetry coming out of the Tx pad to the HC-05. I then tried direct wired connection to my tracker (Tx pad to tracker Rx pin) and Telemetry viewer nor the tracker shows anything going though. I emailed Radiomaster and they said those pads are not Aux 1 and that they did not give the Zorro an Aux 1 so that's confusing if this is working for others. I have the correct settings in Hardware (115200 Baud, Telemetry Mirror). I've tried turning Trainer off/on, I've tried turning off/on the internal ELRS and using the external Ranger, nothing seems to make the pads output anything. I also tried the "EXT TX" pin on the 8pin module connected in case that was it, which also didn't work. 
[image]
What am I doing wrong?

Roman-Golod, 2024-07-25:
Set 57600 or update EdgeTX to the latest version.

dcan999, 2024-07-25:
> Set 57600 or update EdgeTX to the latest version.

I updated to 2.10.2 tonight, going by the above conversations, baud on 2.10.x should be 115k, but I tried 57600 and 400k as well after updating just for hoots, none worked.

Roman-Golod, 2024-07-25:
> > Set 57600 or update EdgeTX to the latest version.
> 
> I updated to 2.10.2 tonight, going by the above conversations, baud on 2.10.x should be 115k, but I tried 57600 and 400k as well after updating just for hoots, none worked. 

Have you read this article? Some settings needed to apply to the bt module.
https://www.aricarc.cl/es/Biblioteca/ZorroMods#bt-mod
Also, if you want to use 3v3, you should replace the module itself from its board, which has stepdown 5v - 3v3.

dcan999, 2024-07-26:
After further clarifying with Radiomaster, they said the pads are indeed AUX1, I guess there was a misunderstanding, but I'm still at a loss for why the tracking isn't seeing anything. I followed these direct-wired instructions exactly ( to rule out bluetooth issues), but the tracker still sees no data. What else could I be doing wrong?

[image]