EdgeTX/edgetx #4808: Branch 2.10 - Trimmers are always displayed

ParkerEde, 2024-03-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The trimmers are now always displayed on the widget pages, regardless of whether you have trimmers on or off in the screen page configuration

### Expected Behavior

The trim displays should be adjustable again

### Steps To Reproduce

1. Create a blank widget page
2. deactivate Trims in page configuration
3. observe, the trim bars are always shown

### Version

Other (Please specify below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

Branch 2.10

pfeerick, 2024-03-25:
I'll have to double check, but I think this is already fixed for RC2

philmoz, 2024-03-25:
My quick fix for #4777 cause this.

PR #4811 (2024-03-26): fix(color): trims are still displayed even when turned off in view settings.
Fixes #4808

Looks like my quick fix for #4777 was not quite right - sorry about that.
closed 2024-03-26 by pfeerick