EdgeTX/edgetx #4813: Allow the last used directory when opening a file to be saved in the Compaion config file

pagrey, 2024-03-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have noticed Companion doesn't write the lastDir entry anymore so File>Open defaults to the main EdgeTX folder. It seems that my entry had carried over from previous config files as I have updated Companion (possibly back as far as OpenTX). Starting with no previous config file the entry isn't created. Manually editing the config file and adding/editing lastDir= works. It would be nice for the lastDir entry to be written so you can easily open files in previous directory. The lastDir entry seems tied to eepromDir but there isn't really an eeprom folder in the settings.

### Describe the solution you'd like

This simple change seems to work for me and lastDir is created and updated in the config file.
https://github.com/EdgeTX/edgetx/compare/main...pagrey:edgetx:save-dir

I'm really just curious why openFile always sets updateLastUsedDir false, as far back as I looked it has been false. I gave up when I started seeing OpenTX. I don't consider this a bug, it seems to work as it is currently written.

pagrey, 2024-03-27:
Curious, the filename is now passed to the openFile method but the line
`g.eepromDir(QFileInfo(fileName).dir().absolutePath());`
is essentially removed because false is alwyas passed to the method. I don't know why the file name had to be passed to the method but since the new method does the same check
`if (!fileName.isEmpty())`
Seems like the fix would be to dump the boolean which doesn't do anything. I think removing it and the associated
`if (updateLastUsedDir) `
would work. The isEmpty already checks that you have a good value to update. I'm not suggesting to change it but that boolean doesn't do anything now other than comment out a single line of code that updates the file path.

I'm going to make the fix on my fork so the boolean is eliminated and the associated check so the eepromDir is updated. Thanks for tracking down the change, I didn't know how to do it.

pfeerick, 2024-03-27:
Using my screenshot as a reference, on GitHub, there is a button up the top that says "Blame" (right under the left pointing arrow). Normally the "Code" buttons is selected. When you hit the "Blame" button, it updates to the view shown there, and you can see when lines were last edited, and jump to the relevant commits. If you want to tunnel back in time further for a particular line, you click on the share icon with a ... trail behind it? ... on that line, and it jumps further back. 

Looking forward to a PR clearing that up ;)

elecpower, 2024-06-11:
@pagrey I see it has been a couple of months since you advised you would submit a PR to fix this. Are you still planning to do so or would you like for me to action?

pagrey, 2024-06-12:
Since it has come up I have created a pull request #5161 with my very simple fix. @elecpower please take any action you would like to, this is just my basic fix for the issue.

elecpower, 2024-10-19:
Save As updates lastDir but not Save

pagrey, 2024-10-19:
As far as I could search updateLastUsedDir is always set to false.

elecpower, 2024-10-20:
> As far as I could search updateLastUsedDir is always set to false.

Just rechecked and you are correct. Thus if not used it should be dropped. I add it to the Save fix #5629.

closed 2024-10-20 by pagrey