EdgeTX/edgetx #4814: Sounds: German is on the SD-card but edgetx is not able to find it.

hannes-m, 2024-03-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I put just the "de" soundpack on the SD-card under "sounds"...wanted to play a track but I was told by edge-tx (2.9.4) "no sounds on the SD-card".
Then I put also the "en" soundpack under "sounds" and it was recognized. 

### Expected Behavior

the "de" should be recognized.

### Steps To Reproduce

Reinstall the SD-card and put just the german sounds on it.

### Version

2.9.4

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

pfeerick, 2024-03-27:
Did you also set your voice language to "de" in settings? 

https://manual.edgetx.org/v/v2.9/edgetx-user-manual/user-manual-for-color-screen-radios/radio-settings/radio-setup

> Voice language - Language for the voice pack. This setting and the voice pack folder on the SD card must match for the sounds to be played.

hannes-m, 2024-03-27:
Sorry...I did not set it...my fault...Thx a lot!Hannes -- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

   

  https://manual.edgetx.org/v/v2.9/edgetx-user-manual/user-manual-for-color-screen-radios/radio-settings/radio-setup

pfeerick, 2024-03-27:
No worries, to hear it's talking to you now! :)

closed 2024-03-27 by pfeerick