EdgeTX/edgetx #4826: FrSky RB-25 lua support

EmanuelStassar, 2024-03-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The only way to setup a FrSky RB-25 at this moment is through Ethos. Flashing the radio to use Ethos for 5 minutes is no fun at all. 

### Describe the solution you'd like

It would be great if the lua script to setup RB-25 could te ported from Ethos to Edge TX.

### Describe alternatives you've considered

Temporary flash Ethos to my X-12s

### Additional context

https://www.frsky-rc.com/rb-25-rb-25s/

STIRLO1980, 2024-03-31:
I am also keen on edgetx lua for FRsky RB25.

pfeerick, 2024-03-31:
Has anyone been in touch with FrSky to ask if they intend to support it for OpenTX/EdgeTX?