EdgeTX/edgetx #4828: [LUA][RC2] Key events constants missing

JimB40, 2024-03-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

EVT_XXX events
----------------------
EVT_PAGE_FIRST 	nil
EVT_PAGE_LONG 	nil
EVT_PAGE_REPT 	nil
EVT_PAGE_BREAK 	nil
EVT_MENU_FIRST 	nil
EVT_MENU_LONG 	nil
EVT_MENU_REPT 	nil
EVT_MENU_BREAK 	nil
EVT_ENTER_FIRST 	1538
EVT_ENTER_LONG 	2050
EVT_ENTER_REPT 	1026
EVT_ENTER_BREAK 	514
EVT_EXIT_FIRST 	nil
EVT_EXIT_LONG 	nil
EVT_EXIT_REPT 	nil
EVT_EXIT_BREAK 	513
EVT_PLUS_FIRST 	nil
EVT_PLUS_LONG 	nil
EVT_PLUS_REPT 	nil
EVT_PLUS_BREAK 	nil
EVT_MINUS_FIRST 	nil
EVT_MINUS_LONG 	nil
EVT_MINUS_REPT 	nil
EVT_MINUS_BREAK 	nil
EVT_ROT_LEFT 	4099
EVT_ROT_RIGHT 	4100
EVT_UP_FIRST 	nil
EVT_UP_LONG 	nil
EVT_UP_REPT 	nil
EVT_UP_BREAK 	nil
EVT_DOWN_FIRST 	nil
EVT_DOWN_LONG 	nil
EVT_DOWN_REPT 	nil
EVT_DOWN_BREAK 	nil
EVT_RIGHT_FIRST 	nil
EVT_RIGHT_LONG 	nil
EVT_RIGHT_REPT 	nil
EVT_RIGHT_BREAK 	nil
EVT_LEFT_FIRST 	nil
EVT_LEFT_LONG 	nil
EVT_LEFT_REPT 	nil
EVT_LEFT_BREAK 	nil
EVT_SHIFT_FIRST 	nil
EVT_SHIFT_LONG 	nil
EVT_SHIFT_REPT 	nil
EVT_SHIFT_BREAK 	nil
EVT_PAGEUP_FIRST 	1540
EVT_PAGEUP_LONG 	2052
EVT_PAGEUP_REPT 	1028
EVT_PAGEUP_BREAK 	516
EVT_PAGEDN_FIRST 	nil
EVT_PAGEDN_LONG 	nil
EVT_PAGEDN_REPT 	nil
EVT_PAGEDN_BREAK 	nil
EVT_SYS_FIRST 	1550
EVT_SYS_LONG 	2062
EVT_SYS_REPT 	1038
EVT_SYS_BREAK 	526
EVT_MODEL_FIRST 	1548
EVT_MODEL_LONG 	2060
EVT_MODEL_REPT 	1036
EVT_MODEL_BREAK 	524
EVT_TELEM_FIRST 	1549
EVT_TELEM_LONG 	2061
EVT_TELEM_REPT 	1037
EVT_TELEM_BREAK 	525
EVT_TOUCH_FIRST 	9844
EVT_TOUCH_BREAK 	8820
EVT_TOUCH_SLIDE 	10356
EVT_TOUCH_TAP 	9332

### Expected Behavior

Those constants are assigned proper evt ID
EVT_EXIT_FIRST 	nil
EVT_EXIT_LONG 	nil
EVT_EXIT_REPT 	nil

EVT_PAGEDN_FIRST 	nil
EVT_PAGEDN_LONG 	nil
EVT_PAGEDN_REPT 	nil
EVT_PAGEDN_BREAK 	nil

### Steps To Reproduce

in LUA
print(EVT_EXIT_FIRST)
or other missing constant name

### Version

Other (Please specify below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### OS Version

2.10 RC2

philmoz, 2024-03-31:
EVT_PAGEDN_xx for radios with both PGUP and PGDN buttons is an oversight. Should be easy to fix.

EVT_EXIT_LONG would never get to a script as it would kill the script first.
Similarly for EVT_EXIT_REPT, by the time it gets to repeating the script would have been killed by EVT_EXIT_LONG.

So looks to me like only EVT_EXIT_FIRST would be missing.

Out of curiosity; wouldn't it be better to use the EVT_VIRTUAL_xxx events in a script?

philmoz, 2024-03-31:
It looks like there was no EVT_EXIT_FIRST event in 2.9.
Is this really needed?
It's just going to add to the FLASH table size.

pfeerick, 2024-03-31:
It looks like it has been a long-standing regression... 2.8 https://github.com/EdgeTX/edgetx/issues/2736 and even 2.6 https://github.com/EdgeTX/edgetx/issues/1631...   ... unless of course it was never present in OTX. :shrug:

3djc, 2024-03-31:
It was never in OTX ;)

pfeerick, 2024-04-01:
In that case, I agree with Phil for now... PAGEDN is the only one needing fixed, as 

```
EVT_EXIT_LONG nil
EVT_EXIT_REPT nil
```

are really invalid since while I think a script actually does see them (as I see two events fire after what would be EXIT_FIRST and EVT_VIRTUAL_EXIT in your devtool), it won't have a chance to actually do much with them as it is being forcefully terminated at the same time and so shouldn't be recommended to be used (IMO, would be better to remove them), and a good use case for `EVT_EXIT_FIRST` to justify it consuming flash memory is needed.

closed 2024-04-01 by pfeerick

JimB40, 2024-04-02:
IMHO constant should be defined for all hardware keys fo consistency.
EXIT (RTN) event numbers are passed to sctipts anyway
What is implemented on top of that (like "stealing" EVT_EXIT_LONG for OS purposes) is another topic.

For example ETHOS implementation passes all events to runNing LUA, then LUA decides if event is "Consumed" by script or passed through to OS.

JimB40, 2024-04-02:
> Out of curiosity; wouldn't it be better to use the EVT_VIRTUAL_xxx events in a script?

In more advanced UI navigation there is need to use shot and long press then EVT_VIRTUAL_xxx thay is usually bound to EVT_xxx_BREAK is not enough.

PR #4829 (2024-04-01): fix(lua): add missing EVT_PAGEDN_xxx events for Lua
Fixes #4828