EdgeTX/edgetx #4836: Prompt requests in new release

trexcarey, 2024-04-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Can the new prompt checklist that can be added as a text file be linked to the switch or action being actioned be asked for?

### Describe the solution you'd like

For example, the request or check item is to move the switch SE to normal mode. Can the system tick the box automatically when the switch is moved to the correct position and once all tasks are complete allow the system to be exited.

3djc, 2024-04-03:
The switch check on start doesn't require checklist and already does what you ask. Checklist are done for thinghs beyond that, like ensuring antenna is in place,....

nikpoz, 2024-04-04:
Hello !!! 
First of all a great thank you to all developeres for their hard work.
I was wondering if a new feature could even be implemented. 
Nowdays many of the esc's come with bluetooth conectivity. It would be great if there was a way to give the radio some access to esc parameters and maybe have it switch among different driving profiles! 
I was thinking it might be possible through a lua script, but at the same time totally unaware if that's even possible.

Many thanks!