EdgeTX/edgetx #4837: Disable trainer function on master movement

LeanderVanhulle, 2024-04-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Trainer function is now enabled/disabled by a switch. As an instructor I'm using the momentary switch on my TX16 and hold this until i need to take over the control. If something happens I'll first need to release (or switch) the switch and only then the trainer function is disabled and my own inputs take over the plane/drone.

### Describe the solution you'd like

I would like an option to disable the trainer function on master stick movement. Especially on the ailerons/elevator sticks.
The trainer function would we switched off until it's enabled again.

### Describe alternatives you've considered

I wrote a whole list of special functions that takes into account the difference of the trainee input and my sticks and compare this to the actual outputs. If there is a difference the trainer function is switched off. But due to the slow calculation the trainer function is also switched off if the student moves it's sticks fast from left to right.

### Additional context

I've not used Spectrum transmitters but i believe these have this function.

AJHAJHAJH, 2024-04-04:
I think close what you want is already achievable.  Perhaps this will help:
https://www.rcgroups.com/forums/showpost.php?p=50716867&postcount=9470

LeanderVanhulle, 2024-04-09:
I've tested the proposed sollution and it indeed seems to work. The only thing is that after deactivating and activating the trainer mode again, you can't disable it by stick movement in the first 3-5 seconds. It seems like the act function isn't triggering again if it was triggered before.  
Second issue i have with the sollution is that the ACT is reacting on any input, not only stick movement. So if i scroll the screen it deactivates, if I for example touch a switch to retract the landing gear, it also deactivates the trainer function.

That's why i would propose a different sollution or have the ACT function split up by for example the act_stick to have only stick movement and not switches or other buttons.

shadowjig, 2024-04-16:
I'd actually like to take this request a step further.  There are two options for trainer input (sticks or channels).  Channels requires that the entire model configuration needs to copied from the master TX to the student TX.  I think that it's completely unnecessary to copy the model config to the student TX.  Furthermore, it may not even be an option if the student TX is a non-EdgeTX/OpenTX capable transmitter.

In my opinion, the best trainer config is stick mode.  Because it removes the need to copy the config to the slave transmitter.  But an issue arises when using sticks mode and setting up a quick takeover via logical switches.

The issue:  Set the master TX to use sticks trainer mode, and setup logical switches to enable a quick take over (i.e. use a physical switch enables trainer mode, and movement on the right stick disables trainer mode).  Once the student moves the sticks on the student TX, the trainer mode on the master TX is disabled (which makes sense).  However, I think the stick movement from the student TX should be inserted after the master TX stick input, but before the master TX mixes are applied. 
 That would allow for a quick takeover function to work when trainer mode on the master TX is set to sticks mode.

Ralf-W, 2024-04-20:
You can set the master TX to Sticks Mode, and ADD the commands from the student Tx sticks.
Now defining a free mixer channel to be the difference between this combined Sticks signal and the signals "TRx" coming from the students sticks. This  difference does create a signal that is purely representing the input from your trainer sticks. Use it to trigger quick takeover:

[image]
[image]
[image]
[image]

I tested it on the bench - and it seems to work as expected.

mha1, 2024-04-20:
creative thinking!

LeanderVanhulle, 2024-04-21:
I tried this solution before and had 2 issues with it: 
1) Adding thrust with add gives the uses only 50% thrust if you set your stick down. Else you really need to hold it still in the center to to interfere with this solution as thrust isn't self centering.
2) if the trainer remote is moving sticks fast from left to right the trainer is deactivated. It seems then that the remote isn't calculating the differences fast enough.

I know you guys try to find an existing solution but they don't work and An easy option should be in the OS.

pfeerick, 2024-04-25:
While complicated, this is probably the best solution I've seen so far for quick takeback

https://youtu.be/030liiCDXYY?si=QME5NQLFTHBY_b9C&t=437

basically you monitor all four sticks for more than a certain amount of *change*, and then you just OR them together until you get a single LS you can use as your 'master wants control' takeover condition. 

However, what is clear from some of issues that have been raised is that there is room for improvement for trainer options, so that might be an area to focus on soon generally.

shadowjig, 2024-04-25:
I appreciate the out of the box thinking and problem solving.  But I agree with @LeanderVanhulle that these features need to be rolled into the OS at this point.  There has been quick take over solutions available for some time now (using logical switches), but it's time to roll that functionality into the OS and reduce the need to create multiple logical switches.

- Separate master stick movement from trainer stick movement (possibly moving trainer input after the master input but before the mixes signal flow)
- Add options into trainer setup to allow for quick takeover.  Allow user to select which stick inputs allow quick takeover.  For example, I might want to use the Ail/Ele stick, but someone else might want to use Rud only as the trigger for quick takeover.
- Add trainer stick movement as options that are selectable as inputs to logical switches (i.e. Ail movement for master vs Ail movement for Trainer)

sande005, 2024-04-27:
All of this, of course, must be an optional configuration.  I shadow fly my students continuously, so when I release the trainer switch my controls are already in the preferred position.  Having only "quick take back" would be detrimental to my ability to assume control.

gagarinlg, 2024-04-27:
> All of this, of course, must be an optional configuration. I shadow fly my students continuously, so when I release the trainer switch my controls are already in the preferred position. Having only "quick take back" would be detrimental to my ability to assume control.

Same here