EdgeTX/edgetx #4839: v2.10.0-RC3 Issue Tracker

pfeerick, 2024-04-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

a

### Expected Behavior

b

### Steps To Reproduce

c

### Version

Other (Please specify below)

### Transmitter

BetaFPV LiteRadio3 Pro

Wandereradi, 2024-04-09:
when we using EdgeTX "Centurion" v2.10.0-RC3 or RC2 on "MT 12" following is observed

1) List of supported protocol displayed is less - i do not see MJX and also the order seems broken from alphabetic order
2) Restore of model saved from eg Edgetx 2.80 is not restored completely in the "Special Functions" - ON Adjust G1 "Source of Input" is not updated but Tickbox is checked.
3) for the MT12 ST trim was T2 and TH trim was T1 in 2.8 but now it is reversed ( RC2 and RC3 )

3djc, 2024-04-09:
1. List of protocol comes from the 4in1 module itself 
2. Mt12 is a 2.10 radio, trying to put older models in there could lead to issues
3. Trims have been purposely changed (https://github.com/EdgeTX/edgetx/issues/4456). Tested it on a new model, works fine.

Wandereradi, 2024-04-09:
> 1. List of protocol comes from the 4in1 module itself

---> I downloaded the 1.3.40 serial AETR build from Multi Protocol org place by selecting the correct Radio but i feel there something is missing coz i only use Kyosho rx but a friend of mine is able to see the MJXq Protocol but i cant. So not sure where what is happening. 

> 2. Mt12 is a 2.10 radio, trying to put older models in there could lead to issues
As per the edge TX webpage the prod built is 2.8 and the website says 2.10 is planned for MT12 so what should i have in the radio. 
[image]

Pencilvestr, 2024-04-17:
> > 1. List of protocol comes from the 4in1 module itself
> 
> ---> I downloaded the 1.3.40 serial AETR build from Multi Protocol org place by selecting the correct Radio but i feel there something is missing coz i only use Kyosho rx but a friend of mine is able to see the MJXq Protocol but i cant. So not sure where what is happening.
> 
> > 2. Mt12 is a 2.10 radio, trying to put older models in there could lead to issues
> >    As per the edge TX webpage the prod built is 2.8 and the website says 2.10 is planned for MT12 so what should i have in the radio.
> >    I am very new to this Edge TX, i have migrated from my stock Kyosho KT331T. I am not trying to make any one mad.
> >    ![image](https://private-user-images.githubusercontent.com/127820040/320827485-7d911535-79a7-40ff-a17e-1a537fe33140.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzMjAwMjgsIm5iZiI6MTcxMzMxOTcyOCwicGF0aCI6Ii8xMjc4MjAwNDAvMzIwODI3NDg1LTdkOTExNTM1LTc5YTctNDBmZi1hMTdlLTFhNTM3ZmUzMzE0MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxN1QwMjA4NDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMWZlMGU3Nzc4MTZhYWNjODZjODQ3MjI5MmIzMjFiMzE2YTU0YWNkNWZmOTE1N2Y4NmZmZjBiZmVjYmRhYmUwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ggb-vDXG88czj57hOW5w0gUAMVD9HCq7IRW573J36QM)

The MPM protocols have been split into surface SFC and Air protocols 
Try Flashing "mm-stm-serial-aetr-sfc-v1.3.4.0"
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/v1.3.4.0

Mick51, 2024-04-19:
In the RC3 version of the Jumper T20S Gemini, the calibration of the potentiometers does not work correctly, even when calibrated there is half available over the entire travel of the potentiometers.

3djc, 2024-04-19:
> In the RC3 version of the Jumper T20S V2 Gemini, the calibration of the potentiometers does not work correctly, even when calibrated there is half available over the entire travel of the potentiometers.

Not seeing that at all. Please delete (after saving) your radio folder a redo calibration

closed 2024-05-04 by pfeerick