EdgeTX/edgetx #4841: widgets error

11ser, 2024-04-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

When upgrading the firmware on the tx16s radiomaster with version 284 to version 2.9.4, the widgets in the f3k (SoarETX ) model stopped working.

### Expected Behavior

I assumed that widgets in existing models of version 2.8.4 would work in version 2.9.4, but they no longer display correctly.

### Steps To Reproduce

We select models in 3k, widgets are displayed further, we scroll through the page with the Page Up key, the widget does not work, we return to the back page psd.  The widget doesn't work.

### Version

2.9.4

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

[image]
[image]
[image]

pfeerick, 2024-04-08:
> I assumed that widgets in existing models of version 2.8.4 would work in version 2.9.4, but they no longer display correctly.

Did you also update the SD card pack? SoarETX is an external set of widgets, so no, you should not be expecting them to 'just work correctly' with newer versions of firmware if there is newer versions of the widgets and sdcard pack available. 

You need a fix that came out after the 2.9.0 sd card pack was released, but I think you should be able to use the [latest](https://github.com/EdgeTX/edgetx-sdcard/releases/tag/latest) or the [2.10-rc1](https://github.com/EdgeTX/edgetx-sdcard/releases/tag/v2.10.0-rc1) sdcard pack with 2.9.x firmware still without issue here. For TX16S you want the `c480x272.zip` zip file.

11ser, 2024-04-08:
> > Я предполагал, что виджеты в существующих моделях версии 2.8.4 будут работать и в версии 2.9.4, но они перестали корректно отображаться.
> 
> Вы также обновили пакет SD-карт? SoarETX — это внешний набор виджетов, поэтому нет, вам не следует ожидать, что они «просто будут работать правильно» с более новыми версиями прошивки, если доступны новые версии виджетов и пакета SD-карт.
> 
> Вам нужно исправление, которое появилось после выпуска пакета SD-карт 2.9.0, но я думаю, что вы сможете без проблем использовать [последнюю версию](https://github.com/EdgeTX/edgetx-sdcard/releases/tag/latest) пакета SD-карт [2.10-rc1](https://github.com/EdgeTX/edgetx-sdcard/releases/tag/v2.10.0-rc1) с прошивкой 2.9.x. Для TX16S вам нужен `c480x272.zip`zip-файл.

yes, I installed the new zip package version 2.9.  Now with package version 2.10 it's working.  Thank you for participating

closed 2024-04-08 by pfeerick