EdgeTX/edgetx #4842: Feature request fro MT12

nikpoz, 2024-04-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No it's just some extra funcionality topic

### Describe the solution you'd like

Hello !!! 
First of all a great thank you to all developers for their hard work.
I was wondering if a new feature could even be implemented. 
Nowdays many of the esc's come with bluetooth conectivity. It would be great if there was a way to give the radio some access to esc parameters and maybe have it switch among different driving profiles! 
I was thinking it might be possible through a lua script, but at the same time totally unaware if that's even possible.

Many thanks!

### Describe alternatives you've considered

Access parameters or esc profiles through the radio using a menu or switch

nikpoz, 2024-04-25:
This is how we do it so far.
But in the drift world, where the is a lead run and a chase run, back to back, there is no time to set all the nessesary parameters... and there might be quite a few to adjust in order to make the car smoother or more agressive. So my thought was, that it would be nice if we could send all those changed parameters with the flick of a switch. That would give the radio an advantage over the rest of the tx/tx systems of the market as well.

3djc, 2024-04-25:
While it sounds interesting, it probably is a can of worms, since I'm pretty sure no two esc use the same (likely undocumented) protocol.

It is an interesting topic for a (user driven) lua implementation tho