EdgeTX/edgetx #4847: [LUA] Add mkdir and rmdir commands

JimB40, 2024-04-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

mkdir - creating directory (folder)
and 
rmdir - removing directory (folder) -possibly recursive for sub folders and files 
are missing commands to make LUA apps installers happen.

Considering number of how-to install videos this the way to go :)

### Describe the solution you'd like

Just add.

philmoz, 2024-04-09:
'rmdir' functionality already exists - https://luadoc.edgetx.org/part_iii_-_opentx_lua_api_reference/filesystem/del-file-or-directory

philmoz, 2024-04-10:
The code for the 'del(name)' function does not match the description in the EdgeTX Lua documentation.

If the delete function fails for any reason it returns no values at all - the documentation says it should return the F_RESULT of the operation. It will only return the F_RESULT if the delete succeeds in which case it will always be 0 (FR_OK).

Do you want to fix the code or the documentation?

JimB40, 2024-04-10:
> If the delete function fails for any reason it returns no values at all - the documentation says it should return the FRESULT of the operation. It will only return the FRESULT if the delete succeeds in which case it will always be 0 (FR_OK).

Error handling when deleting is important. I'd rather fix code.
Is it due to?
1. Bug in implementation
2. Striiped functionality when condiug?

is del(folderName) recursive?
Does it  delete all sub-folders and files in those folders?

philmoz, 2024-04-10:
> Error handling when deleting is important. I'd rather fix code. Is it due to?
> 
> 1. Bug in implementation
> 2. Striiped functionality when condiug?

Looks like it was changed in PR #4288 by mistake.
 
> is del(folderName) recursive? Does it delete all sub-folders and files in those folders?

No, the folder must be empty first.

JimB40, 2024-04-11:
> No, the folder must be empty first.

Hard to make rmtree(folderNname) or del(file, tree)?
If so it can done via recursive LUA function using fstat() & del()

philmoz, 2024-04-11:
> Hard to make rmtree(folderNname) or del(file, tree)? If so it can done via recursive LUA function using fstat() & del()

Not hard; but given the limited use it might be simpler to do in Lua.

pfeerick, 2024-04-12:
Since it is possible do in Lua already by using two/three functions together, better to do it that way, and instead do loadable/includable Lua utility function libraries... less flash and RAM usage... especially if you don't use those functions (or Lua scripts at all).

philmoz, 2024-04-12:
```Lua
local function rmdir(path)
  local stat = fstat(path)
  if bit32.band(stat.attrib,AM_DIR) == AM_DIR then
    for nm in dir(path) do
      rmdir(path..'/'..nm)
    end
  end
  return del(path)
end
```

JimB40, 2024-04-16:
Fine with me. Let's LUA do it.

Just realized we we are missing copy or move (copy & del) function to have tools for instaler.
Idea is that user will put app install folder in ROOT or /INSTALL folder then LUA script will move files to proper locations and delete app install folder after.

Best would be move dir or file but copy() with existing del() & rmdir() will also do the job.

we do file distribution via
- dir(source) -> mkdir(target)  in case od folders
- io.read(source) -> - io.write(destination)
- del(source) - to clean up
but it seems little cumbersome comparing to simple move()

philmoz, 2024-04-18:
I added a 'rename(from_path, to_path)' function. It will move files or directories.

Note: If you are moving a file or directory to a new parent, then the new parent location must already exist.
E.G.
    rename("/INSTALL/file1.lua", "/WIDGETS/Widget/file1.lua")
will fail if the "/WIDGETS/Widget" directory does not exist.

PR #4870 (2024-04-28): feat(lua): add 'mkdir'  and 'rename' functions
Resolves #4847

Also fixes bug introduced in https://github.com/EdgeTX/edgetx/pull/4288
closed 2024-04-28 by pfeerick