EdgeTX/edgetx #4852: Hat Mode not Working on EL18 on ETX v2.10 RC3

avella007, 2024-04-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

I would like to share some information  to the EdgeTX developers.  

I just upgraded my FlySky Elysium EL18 to the latest EdgeTX firmware (v2.10 RC3) including updating the SD card.  

I tested out the Hat Mode and I set it to every options available and it just doesn’t work.  I found that I’m only able to use the Trim buttons when scrolling through the menus of a Lua Script or the TBS Agent lite.  

When using the EdgeTX menus, I’m only able to select a menu item (using the right trim) and exit out of the current menu (using the left trim) other than that they function as trims.

### Expected Behavior

The Trims are just not working as they should be when the Hats Mode is set to either the Keys Only or the Switchable Mode

### Steps To Reproduce

1. Set the Hats Mode to either Keys Only or the Switchable Mode
2. Exit out of the Radio Settings
3. Start using the Trim keys to access the EdgeTX menus and you will see that you anre able to use the Trim buttons when scrolling through the menus of a Lua Script or the TBS Agent lite.  
4. When using the EdgeTX menus, I’m only able to select a menu item (using the right trim) and exit out of the current menu (using the left trim) other than that they function as trims.

### Version

Other (Please specify below)

### Transmitter

Flysky EL18

philmoz, 2024-04-07:
Did you also set the 'Hats mode' on the model (Model Setup -> Trims -> Hats mode)?

pfeerick, 2024-04-07:
ie. Check that that `Trims` -> `Hats Mode` is set to `Global` (unless you want to override the setting for a given model regardless of the global radio settings) ... it _should_ default to that but it is possible it didn't for you. Navigation with the trim hats is working for me on EL18 with 2.10-RC3

avella007, 2024-04-08:
> ie. Check that that `Trims` -> `Hats Mode` is set to `Global` (unless you want to override the setting for a given model regardless of the global radio settings) ... it _should_ default to that but [...]

Will confirm this evening, but it may be that my Trims are not set to Hats mode.  

Thank you.

avella007, 2024-04-08:
All is good.  I apologize for my ignorance.   I had not set the Trims to Hats mode at model level.  

Please close this issue, since it is not an issue, but rather than a user error.  

Thank you again to all those who helped me out.

closed 2024-04-08 by avella007