EdgeTX/edgetx #4855: Start up splash screen delay broken

Kevltan, 2024-04-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Start up splash screen delay seems to be broken in the latest nightly

### Expected Behavior

Splash screen delay up tp 15s

### Steps To Reproduce

Start up with splash delay entered

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster TX16S / TX16SMK2

philmoz, 2024-04-07:
It works on my TX16S - what exactly is the problem?

pfeerick, 2024-04-07:
> Start up splash screen (including sound file) delay seems to be broken in the latest nightly. 

Delay possibly being skipped by starup errors i.e. switch, stick warnings. If there are no startup errors, it waits approx 15 seconds for me. 

> Also shutdown theme now defaults to black and not as set - has field or parameters changed for this?

Confirmed... it no longer follows the theme colours as intended.

Kevltan, 2024-04-08:
On my previous nightly 2.11 the startup delay would be the set time
regardless of switch positions or warnings - something has changed.
I agree that when there are no active alarms there is the set delay time,
but is this right and why has it changed?

PR #4865 (2024-04-09): fix(color): restore previous startup and shutdown functionality.
Wait for splash screen to finish before showing throttle and switch alerts.
Use theme color for shutdown animation background.

Fixes #4855
closed 2024-04-09 by pfeerick