EdgeTX/edgetx #4859: Jumper t16 issues with EdgeTX "Centurion" v2.10.0-RC3

beach97, 2024-04-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Flysky digital hall gimbal cannot work on T16，Unable to use even after recalibration.

### Expected Behavior

I hope to use it normally and really want to experience the new features brought by 2.10

### Steps To Reproduce

Upgrade to 2.10rc using Flysky digital hall gimbal Joint T16

### Version

Other (Please specify below)

### Transmitter

Jumper T16

### Operating System (OS)

Windows

pfeerick, 2024-04-08:
> Unable to use even after recalibration.

What does this mean? Do the gimbals not work at all? Intermittent response? Poor resolution?

beach97, 2024-04-08:
https://github.com/EdgeTX/edgetx/assets/39640832/32ac9c87-d6ef-474b-bcfb-f120f2d26bb5

> gimbals

Yes, the gimbals doesn't work at all

beach97, 2024-04-08:
> > Unable to use even after recalibration.
> 
> What does this mean? Do the gimbals not work at all? Intermittent response? Poor resolution?

Yes, the gimbals doesn't work at all

stefan73, 2024-08-12:
I got exactly the same problem on my Jumper T16. I tried with 2.10.3. Going back to 2.9.X makes the gimbals work again. Calibration on 2.10 is impossible. 
According to discord support I also updated the bootloader to 2.10.3 by DFU flash. I also deleted the models and radio folders and rebooted the radio. Then it goes over storage prep and jumps immediately into calibration. No success either.
Happy to help logging. I do not have HW debug capabilities.

stefan73, 2024-09-13:
Bug continues to exist in 2.10.4

stefan73, 2025-04-27:
Bug continues to exist in 2.10.6

stefan73, 2025-04-27:
The gimbals work, when I flash the TX16s firmware. As several buttons are misaligned, this is not a real fix. But it may help in finding the bug.

stefan73, 2025-04-27:
OK, now I need some help. I found the root cause in 2.10. in src/targets/horus/hal.h 
For the TX16s there are no PWM gimbals. But for the T16 there are.

It seems that the code has a double definition of the HW used:

`#if !defined(RADIO_TX16S) && !defined(RADIO_T15) && !defined(RADIO_F16) 
    #define PWM_STICKS
    #define PWM_TIMER                   TIM5
    #define PWM_GPIO                    GPIOA
    #define PWM_GPIO_AF                 GPIO_AF_TIM5
    #define PWM_IRQHandler              TIM5_IRQHandler
    #define PWM_IRQn                    TIM5_IRQn
    #define PWM_GPIOA_PINS              (ADC_GPIO_PIN_STICK_LH | ADC_GPIO_PIN_STICK_LV | ADC_GPIO_PIN_STICK_RH | ADC_GPIO_PIN_STICK_RV)
    #define STICK_PWM_CHANNEL_LH          0
    #define STICK_PWM_CHANNEL_LV          1
    #define STICK_PWM_CHANNEL_RV          3
    #define STICK_PWM_CHANNEL_RH          2
  #endif`

and here are the FLYSKY_GIMBAL definitions:
`// FlySky Hall Sticks
  #define FLYSKY_HALL_SERIAL_USART                 UART4
  #define FLYSKY_HALL_SERIAL_GPIO                  GPIOA
  #define FLYSKY_HALL_DMA_Channel                  LL_DMA_CHANNEL_4
  #define FLYSKY_HALL_SERIAL_TX_GPIO_PIN           LL_GPIO_PIN_0  // PA.00
  #define FLYSKY_HALL_SERIAL_RX_GPIO_PIN           LL_GPIO_PIN_1  // PA.01
  #define FLYSKY_HALL_SERIAL_GPIO_AF               LL_GPIO_AF_8
  #define FLYSKY_HALL_SERIAL_USART_IRQHandler      UART4_IRQHandler
  #define FLYSKY_HALL_SERIAL_USART_IRQn            UART4_IRQn
  #define FLYSKY_HALL_SERIAL_DMA                   DMA1
  #define FLYSKY_HALL_DMA_Stream_RX                LL_DMA_STREAM_2`

If the eliminate the PWM_STICKS code from the compilation (e.g. by adding && !defined(FLYSKY_GIMBAL)) the gimbals work.

Now how should this be fixed best? - Check for Flysky Gimbals before instanciating the PWM Gimbals?

gagarinlg, 2025-04-27:
> Now how should this be fixed best? - Check for Flysky Gimbals before instanciating the PWM Gimbals?

Yes, exactly that. This is also what we do with analog gimbals, that are connected to the ADC. First look for Flysky gimbals, if there are none, initialise the analog gimbals.