EdgeTX/edgetx #4873: Jumper T12 Pro will not power on

gherman59, 2024-04-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When T12 Pro power button depressed, start up bars appear on screen but goes dark after 2 seconds and transmitter does not turn on.  AM able to power on via boot loader. Powering on to boot loader appears normal and can exit boot loader to the power om status, but get constant blue power light. 

### Expected Behavior

Expect power on count up screen, welcome message and green power light.

### Steps To Reproduce

1. Press power and hold button for 2 seconds.  
2. Release when four bar count up screen completes.  
3.  Hear welcome message. 
4. See green power light. 

### Version

Other (Please specify below)

### Transmitter

Jumper T12, Jumper T-Lite

### Operating System (OS)

Windows

### OS Version

N/A

### Anything else?

Loaded 2.10.0 RC 3.

hgo-be, 2024-07-22:
After migration from Opentx to EdgeTx 2.10.2 I have the same behavior. After a lot of trials the T12 sometimes starts up. Problematic in the 2.10.x versions. 
I found a work around by setting Haptic Mode to Quiet.

upya4ko, 2024-11-14:
> I found a work around by setting Haptic Mode tor Quiet.

Confirm, have same problem (t12).
Options->Radio Setup (3/6)->**Pwr ON/OFF Haptic** - **remove check-mark**
And now PwrON work, and haptic work for alarms etc.

Seems problem in start checks or requirements in use haptic lib on power on,  battery is OK, on OpenTX problem absent,

pfeerick, 2025-02-07:
Is this still the case with a currently nightly? All I can think of is maybe since haptic is clearly working otherwise, maybe it is being triggered at a time the radio power supply isn't ready for it... #5017 (2.10) added the *option* to have haptic on power on, but #3478 (2.9) had it on without option to disable... prior to that (ie. 2.8) there wouldn't have been any haptic on power on. So without any feedback as to issues with those versions, it's hard it pin down exactly what is causing it.

upya4ko, 2025-02-14:
> maybe it is being triggered at a time the radio power supply isn't ready for it...
Before: **2.11.0-nightly** git -  c428011e  - BUG EXIST
It make sense, i last days plug-unplug battery, and every time **first power ON** can do only thru **"bootloader > exit"**, after it all work OK (in settings power on **haptic disabled**)
 
TX - Jumper T12

Test: **t12-f51c67a_2.11RC1.bin** - BUG FIXED
Haptic ON
Work 1 of 10 try, hold button as usual, can be clear screen or 2-3 squares and off, one time count full scale and VIBRATE (i release button) boot OK.
So it works in 1 of 10 try
3 times count to 3 "squares" and off, on 4 try all 4 "squeres" and haptic + boot FW.

In my opinion problem may be in timings (time to wait button pressed before enable haptic.) i set "Pwr On delay: 1s"

Continue testing, replug battery and it again 5 times fail on 5 powerON normally with haptic!
**Now it power ON 9/10 times** (like before bug, sometimes it not start from first push).
Now i can again power TX by "haptic"

> Is this still the case with a currently nightly / v2.11-rc? All I can think of is maybe since haptic is clearly working otherwise, maybe it is being triggered at a time the radio power supply isn't ready for it... [#5017](https://github.com/EdgeTX/edgetx/pull/5017) (2.10) added the _option_ to have haptic on power on, but [#3478](https://github.com/EdgeTX/edgetx/pull/3478) (2.9) had it on without option to disable... prior to that (ie. 2.8) there wouldn't have been any haptic on power on. So without any feedback as to issues with those versions, it's hard it pin down exactly what is causing it.

So in my case new **FW 2.11RC1 fix issue** to prevent first power on thru "bootloader" .

gherman59, 2025-02-15:
I am still have the power on problems with the T12 Pro, and have tried running FW 2.11.0 RC1.  Vibration function does not seem to impact the problem - have tried with it set normally on and off. The Problem remains the same. T12 goes to boot loader correctly, and from boot loader if I exit the t12 powers on and operates correctly from there.  The T12 will not power on if done by pushing on button alone for 4 seconds as it use to do.  I have also noticed that during power on attempts, the power light illuminates green during start up, and when radio completes power on via boot loader the power light illuminates blue.  This is backwards of what the T12 use to do.

upya4ko, 2025-02-15:
> I am still have the power on problems with the T12 Pro, and have tried running FW 2.11.0 RC1. Vibration function does not seem to impact the problem - have tried with it set normally on and off. The Problem remains the same. T12 goes to boot loader correctly, and from boot loader if I exit the t12 powers on and operates correctly from there. The T12 will not power on if done by pushing on button alone for 4 seconds as it use to do. I have also noticed that during power on attempts, the power light illuminates green during start up, and when radio completes power on via boot loader the power light illuminates blue. This is backwards of what the T12 use to do.
> 
> Might I have a hardware problem with the power on switch and latching circuit, or does this sound like firmware problem?

 It firmware problem 100%, my TX in good shape, on OpenTX this "Bug" is absent, after migrate to EdgeTX it "broke" i think problem in bootloader, revert to OpenTX, all OK but lack of support.
So i find "Solution" disable vibrate on start and it give acceptable power on sequence, sometimes it "Off" on 3 "square" , some time only blink backlight, but from 3-4 try it power on and work good, after update to 2.11.0 RC1 i enable back haptic on power on (i like power on TX when take it in hand, so it already on when i see screen, vibration say "Ok, release button").

pfeerick, 2025-02-15:
Have you both ensured you are running the same bootloader AND firmware version (i.e. by updating via DFU - which does both, or explicitly updating firmware from bootloader, and bootloader from firmware)?

upya4ko, 2025-02-15:
> explicitly updating firmware from bootloader,
 
 I flash only FW (i not fully understand how flash bootloader,  on migration to EdgeTX i flash "transis" bootloader and make many different combination to restore bootloader, now installed bootloader: 2.11.0-nigtly (f5ic67af)

gherman59, 2025-02-15:
I am using firmware and boot loader v2.11.0 RC1.  Have tried every version from v2.9.4. 
Only difference I have found is on v2.10.2 and older, the power on light colors sequence was correct.

pfeerick, 2025-02-15:
Power light _sequence_ is probably unrelated... it was deliberately changed to blue "booting", green "ready" for all handsets where RGB power indication was an option.

pfeerick, 2025-02-16:
>now installed bootloader: 2.11.0-nigtly (f5ic67af)).
> 
> So if problem in bootloader where get latest ? only using "companion" ?

If you use either [EdgeTX Buddy](https://buddy.edgetx.org/) to flash in USB mode or Companion, bootloader and firmware are updated in one step via DFU.

If you are updating via writing the firmware to the SD card, then using the bootloader to update the firmware, you can  boot into the firmware, go into SYS -> SD -> FIRMWARE, select that firmware file, and choose the "Flash Bootloader" option from the popup menu to install the matching bootloader. 

> i see "contrast and brightness" not take effect on bootloader

No, because none of the SD card settings are read by the bootloader. Brightness and contrast values are "hard-coded".

upya4ko, 2025-02-16:
> FIRMWARE, select that firmware file, and choose the "Flash Bootloader"
I understand right i need flash SAME v2.11-rc.bin with FW or need download other BIN file ?
I try many things so not remember what works.

For now i have problems with VM that use for "Companion", it see "Transis bootloader" and "no DFU found", it maybe broke because i "Inject driver" to restore "SoftBrick", now i use "Manual thru SDcard" .

I now try connect TX to VM and it found "Damaged FS on SDcard" i choose repair and save.
(now try repair VM or run Buddy inside Docker).

Interesting... i use "Cheep 4GB" SD card, i have some different SDcards, if bootloader search "Press time" on SDcard maybe problem in card? (some time it starts with "No SDcard" message.

Now i make couple cards and watch result (all content same).

Question about flashing bootloader is still active, do not want again "deBrick" TX...

pfeerick, 2025-02-16:
> For now i have problems with VM that use for "Companion", it see "Transis bootloader" and "no DFU found", it maybe broke because i "Inject driver" to restore "SoftBrick", now i use "Manual thru SDcard" .

This sounds more like you are not flashing in DFU mode. You leave the handset powered off for DFU mode. It is not possible to brick with DFU mode, the code for this is part of the MCUs internal ROM. 

> Interesting... i use "Cheep 4GB" SD card, i have some different SDcards, if bootloader search "Press time" on SDcard maybe problem in card? (some time it starts with "No SDcard" message.

This can indeed cause issues, as some cheaper SD cards don't init properly in the required / spec time period. It has became more an issue with later versions of EdgeTX, presumably because we are less tolerant of non-standards compliant cards.

upya4ko, 2025-02-16:
>  You leave the handset powered off for DFU mode. It is not possible to brick with DFU mode, the code for this is part of the MCUs internal ROM.

You are right, i just forget (read too many manuals about MCU and lack of sleep).
Just "Plug" and see "STM32  BOOTLOADER" thank you!

---

> cheaper SD cards don't init properly in the required / spec

May be a true problem, so i have time and hardware to test this theory.

## SDcards:

_I download clean **EdgeTX "Centurion" v2.10.5 SD Card Pack + Glados sound pack :) )** card and format all cards "Fresh" arter ```/dev/zero```_
_I use integrated hight speed card-reader_

### First Update bootloader and FW

__Copy t12-f51c67a_2.11RC1.bin__
__Clean contacts using IPA__
__Boot FW, update bootloader__

### Prepeare cards

1. DiscDestroy :)
2. MBR 1 partition Fat32 0%-100% primary
3. Copy "Fresh install"
4. Init card in TX, and leave "DEFAULTS"
5. Test for "BUG"

 1. Kingston __512MB__
  * __BUG PRESENT!__
 2. NoName __2GB__
  * 10 times fail on 11 "No SD card"
  * __BUG PRESENT!__
 3. SiliconPower  __4GB__ __Class 4__(This i use for now)
  * After DD ```task sync:231402 blocked for more than 1208 seconds.``` wow... 
    ```(4.0 GB, 3.7 GiB) copied, 1466.56 s, 2.7 MB/s```
  * __BUG PRESENT!__
 4. GoodRam __32GB__ __U1__ __Class 10__
  * powerON 1 of 10 tryes
  * __BUG PRESENT!__
 5. SanDisk __32GB__ __U1__ __Class 10__
  * __Total FAIL__ (not power on, in bootloader "No SD card"
 6. SanDisk __64GB__ __U1__ __Class 10__
  * 3/10 power on
  * __BUG PRESENT!__

## Same test BUT use only first 256MB of each SDcard.

1. DiscDestroy :)
2. MBR 1 partition Fat32 0%-256MB primary
3. Copy "Fresh install"
4. Init card in TX, and leave "DEFAULTS"
5. Test for "BUG"

 1. Kingston __512MB - 256MB__
  * 5/15 powerON
  * __BUG PRESENT!__
 2. NoName __2GB - 256MB__
  * 0/10
  * TOTAL FAIL
 3. SiliconPower  __4GB - 256MB__ __Class 4__(This i use for now)
  * Power On 3/10 on 11 "No SD card"
  * TOTAL FAIL
 4. GoodRam __32GB - 256MB__ __U1__ __Class 10__
  * 1/10 powerON
  * TOTAL FAIL
 5. SanDisk __32GB - 256MB__ __U1__ __Class 10__
  * 0/10
  * TOTAL FAIL
 6. SanDisk __64GB - 256MB__ __U1__ __Class 10__
  * 1/10
  * TOTAL FAIL

---

Need think about it, i find smallest (4GB) 10 class card, like i think less size - better.

For example on PSVita to use sd2vita need format SDcard to exfat without part table and with cluster size 64K, it driver requairements.

Maybe need some "special" prepare for SDcard?

---

I test throry, need think, maybe buy smallest 10 class SDcard.
IMHO SDcard changes corelate with "bug".
Maybe CPU not like "too small" or "too big" clusters?

pfeerick, 2025-02-16:
We usually recommend using the SD Card Association formatter when trying to rule out SD card formatting issues... https://www.sdcard.org/downloads/formatter/

But usually any formatting tool that formats to FAT16/FAT32 standard should be fine. There has been some recommendation to use 4K sector sizes, as it appears to help with audio not being choppy. 

It unfortunately sounds either we have something misconfigured for the T12, if it can work perfectly fine under OTX but has issues with ETX. It would be very interesting if you were to try [v2.5.0](https://github.com/EdgeTX/edgetx/releases/tag/v2.5.0) and [v2.6.0](https://github.com/EdgeTX/edgetx/releases/tag/v2.6.0). v2.5.0 still used the EEPROM for model and radio settings storage, and v2.6.0 was the first version to move B&W over to SD card storage, and is from well any of the timing changes... If 2.5.0 works fine, it could be an issue with T12 and SD cards generally. If 2.6.0 works fine, it could be the timing changes are the problem (which seems unlikely given your latest tests, but I will go where the data leads me 🤭 )

upya4ko, 2025-02-16:
> [v2.5.0](https://github.com/EdgeTX/edgetx/releases/tag/v2.5.0) and [v2.6.0](https://github.com/EdgeTX/edgetx/releases/tag/v2.6.0). v2.5.0

OK, i make "tests" and try "latest OpenTX" to check again, it take some time, update soon.

gherman59, 2025-02-16:
I will also try OpenTX, Edge v2.50, Edge v2.6.0 to compare power on behavior.  Will be a few days before I can complete.

upya4ko, 2025-02-16:
# So here tests

_My models already f\*p by migrating to EdgeTX so i can "Wipe TX" with no problem_
_Also i have "audio stutters" on power on or use "logical switches", so maybe this part of issue too_

## Test setup:

- Jumper T12
- MicroSD Kingston 512MB Sector 512 MBR + Prim. part.
- All versions "stock card" same version with sound (en) to make realistic setup.
- All flash over SDcard (bootloader+FW), card every time DD\>format\>copy for "clean result"

## Test script

```sh
#!/bin/sh
# i am lazy, if action need repeat more 2 times write script)

sdCard=$1
mountPath='/mnt/sdcard'
storagePath=$2

read -p "Start process ${storagePath} to ${sdCard} ? " userAnsw

case ${userAnsw} in
  "y" | "Y" | "yes" | "YES")
    echo "Start in 5..." ; sleep 1
    echo "Start in 4..." ; sleep 1
    echo "Start in 3..." ; sleep 1
    echo "Start in 2..." ; sleep 1
    echo "Start in 1..." ; sleep 1

    echo "Start!"
    sudo dd if=/dev/zero of=${sdCard} bs=1M status=progress
    sudo parted ${sdCard} mktable msdos
    sudo parted ${sdCard} mkpart primary fat32 0% 100%
    sudo mkfs.msdos -n EdgeTX -S 512 ${sdCard}1
    sudo mount -t vfat -o rw,umask=000 ${sdCard}1 ${mountPath}
    cp -rv ${storagePath}/* ${mountPath}/
    sync
    echo 3 | sudo tee /proc/sys/vm/drop_caches
    sudo umount ${sdCard}1
    sync
    echo 3 | sudo tee /proc/sys/vm/drop_caches

    echo "Write ${storagePath} to ${sdCard} DONE!"

    exit 0
  ;;
  *)
    echo "Cancel!"
    exit 1
  ;;
esac
```

## Test subj.

- OpenTX
  - 2.3.15 - opentx-t12-faichoice-flexr9m-internalmulti-lua-noheli-sqt5font-en-2.3.15-otx.bin
- EdgeTX
  - 2.4.0 - t12-09a6580.bin
  - 2.5.0 - t12-28cdb40.bin
  - 2.6.0 - t12-338d4d2.bin
  - 2.9.0 - t12-0ff73cc.bin
  - 2.10.0 - t12-22ce06d.bin 
  - nightly-2f1ede37 (2025-02-16) - t12-2f1ede3.bin

```
./EdgeTX_2.10.0:
edgetx.sdcard.version  FIRMWARE  LOGS  MODELS  RADIO  SCREENSHOTS  SCRIPTS  SOUNDS

./EdgeTX_2.11.NIGHTLY:
edgetx.sdcard.version  FIRMWARE  LOGS  MODELS  RADIO  SCREENSHOTS  SCRIPTS  SOUNDS

./EdgeTX_2.11.RC1:
edgetx.sdcard.version  FIRMWARE  LOGS  MODELS  RADIO  SCREENSHOTS  SCRIPTS  SOUNDS

./EdgeTX_2.4.0:
edgetx.sdcard.version  FIRMWARE  LOGS  SCREENSHOTS  SOUNDS

./EdgeTX_2.5.0:
edgetx.sdcard.version  FIRMWARE  LOGS  SCREENSHOTS  SOUNDS

./EdgeTX_2.6.0:
edgetx.sdcard.version  FIRMWARE  LOGS  SCREENSHOTS  SOUNDS

./EdgeTX_2.9.0:
edgetx.sdcard.version  FIRMWARE  LOGS  MODELS  RADIO  SCREENSHOTS  SCRIPTS  SOUNDS

./OpenTX_2.3.15_2022-04-22:
EEPROM  FIRMWARE  LOGS  MODELS  opentx.sdcard.version  SCREENSHOTS  SCRIPTS  SOUNDS  UTILITIES

```

```
cat ./*/*.sdcard.version
2.10
2.10
2.10
2.5
2.5
2.6
2.9
2.3V0039
```

## Results

* OpenTX
  - 2.3.15
    - HaptON - 5/5 __NO BUG__ (But all my models come back... weird, maybe not clean EEPROM? )
    - HaptOFF - Not tested
* EdgeTX
  - 2.4.0
    - HaptON - 5/5 __NO BUG__ ("Bad eeprom data, formatting EEPROM" in "7/7" EEPR: 220) 
    - HaptOFF - Not tested
  - 2.5.0 
    - HaptON - 5/5 __NO BUG__ ("Models clean")
    - HaptOFF - Not tested
  - 2.6.0 
    - HaptON - 5/5 __NO BUG__ (Warning - EEprom Data v220 - converting)
    - HaptOFF - Not tested
  - 2.9.0 
    - HaptON - 5/5 __NO BUG__
    - HaptOFF - Not tested
  - 2.10.0
    - HaptON - __1/6__ __BUG PRESENT!!__ (Here LED color swaped)
    - HaptOFF - __2/15__ __BUG PRESENT!!__
  - 2.11.RC1
    - HaptON - __0/5__ __BUG PRESENT!!__
    - HaptOFF - __0/5__ __BUG PRESENT!!__
  - nightly-2025-02-16 (3.0.0)
    - HaptON - __0/10__ __BUG PRESENT!!__
    - HaptOFF - __2/15__ __BUG PRESENT!!__

---

## In summ

Problem appear between 2.9.0 and 2.10.0
Need search what changed.
For now install 2.9.0 and wait, if need test some builds i can help.
Name "T12BlueBug" may correlate with change RGB led)

gherman59, 2025-02-17:
Tested my T12 Pro.  Had similar results.  With 2.9, radio powers on normally and has normal RGB light operation and normal audio.  With 2.10, it fails to power on except if powered on via the bootloader, the RGB green and blue are reversed, and the audio is chopped sounding.  

Agree with "upya4ko", problem appears between 2.9.0 and 2.10.0.

I am leaving my T12 Pro on 2.9.0 for now.  Works well enough as a student radio. 

Will test as bug fixes or upgrades are released.

3djc, 2025-02-17:
@upya4ko ,could you test with this T12 firmware ?

[firmware.zip](https://github.com/user-attachments/files/18821872/firmware.zip)

upya4ko, 2025-02-17:
Hi  @3djc, flash and test, flash test build 1, after flash 3 tries **power ON OK**, for more real situation re-plug battery and test   **15 / 20**  POWER ON!
Bug not disappear fully.  

Looks like "Nice shoot", found one interesting correlation, if before "Power On Button" wait 3s+ chance to start chance 70%, if not wait 2-4 "squares" and not power ON.

Now start from 2-3 try:
1. 4 "squares" and off,  
2.  back-light blink off.
3. Normal power ON

Set "Pwer On delay: 1S
13/20 power ON, so now chance around 70% to first start.

In more "real usage" i write this text and parallel test with  pause 50+sec, not correlate with "wait time" even 2m give start from 3 try, other try from first time.
---

In conclusion, situation make better but not fully, for now TX power ON 7/10 times, so between  2.10.0 and today test build BIG difference.
On 2.10.0 - chance 50%
On today test build 1 - chance 75-80%

Now TX power on from 1-3 try with HapticON.

Thank you for build, it more usable than RC1 for T12.

Thanks, if need more test i am here)

upya4ko, 2025-02-17:
@gherman59 Please test on tour TX, maybe my SDcard is failing soon, now i test and "PowerON TX" іn big intervals, and EVERY TIME it START OK!
So maybe bug fixed and i only need new SDcard)

gherman59, 2025-02-17:
@upya4ko upya4ko

Tested the shared firmware.  T12 had normal power on 2 of 10 tries.  Power on via boot loader works every time.  RGD blue and green still reversed.  I was getting 0 of 10 with 2.10.   T12 on the failed attempts does not complete the power on sequence, I get 1 to 3 bars, hear a little click from speaker and radio turns off. 

By the way, I am using Gigastone FAT32 8gb SD card in the T12 and using the same SD in three other radios, no troubles.  Have had poor performance if I use larger than 8gb in the T12, and in the TX16s MKII where radio would powers on okay but would take 45 seconds to complete power on sequence.

upya4ko, 2025-02-17:
@gherman59  maybe size of SDcard is issue too, if micro-controller (esp32) be read SD card it may be very slow, and if "Card too big" it can overfill "read timout" i order 4GB 10class (not found smaller) for TX and test more.

PS. 5/5 PowerON (10-20m between button push)

upya4ko, 2025-02-17:
@gherman59  you can test SD card "speed" using simple tool

upya4ko, 2025-02-17:
@gherman59  thank you for tests.
> I am guessing that is good enough?

Yes,10 class is 10+MB write (read faster), better only less capacity, i found 2GB 10 class card (now f*k seller to force seller test product... "Manufacture say 10 class" i say "Ok, return for your cost)").  
In my case with 512MB card 3-5 class all work as OpenTX, My TX lay on table and i time to time "power on" for testing, for now 10 of 10 power ON OK.

Maybe need try create 1 partition 512 MB ?  (not remember where get it)
In my case old 512MB card work better than 4GB 4 class.

IF MCU read ONLY 1 partition it can help to reduce "Red time" for small partition.

gherman59, 2025-02-17:
Mine is still doing about 2 of 10 with the testing firmware, even if I wait several minutes before attempts.  I have some ScanDisk FAT32 4gb micro SD in my GoPro camera stuff.  Will try one of those in the next few days.  From what is on the SD card, a smallish sized one should work fine.

upya4ko, 2025-02-18:
Update:

I half of a day "testing" randomly push button when walk around, i not found pattern.
20/30 Power ON
```
111111100111001100111011011100
```

Totally random.

gherman59, 2025-02-19:
@upya4ko upya4ko

Tried a new and tested as good FAT32 4gd microSD (ScanDisk HC/C4). 
T12 behavior is the same.  Power on via Bootloader works, normal power on with power button fails. This time had 1 of 20 power on attempts work.  That is the smallest I have available.

gherman59, 2025-02-23:
@upya4ko and all.

Want to make sure that SD card size is not the cause of the failure to turn off troubles with my T12Pro. 
Found and ordered a couple FAT32 1gb MicroSD (Lerdisk HC/C4).  Received and cycle tested them with acceptable results, write speed 7.1 mbs / read speed 57 mbs.  Loaded the the SD card and gave it a try and have the same failure to power on symptoms as with the 4gb and 8gb SD cards.   
With EgheTX v2.9.4 my T12Pro powers on and operates normally.  
With EdgeTx v2.10 and newer, the power on fault occurs. Have tried all version up to the testing copy of firmware provided a few entries ago in this thread.  The T12Pro can be powered on via the bootloader with v2.10/v2.11.  

In my case it looks like SD card size is not the problem.

Willing to load and test as needed.

My best,
GH

upya4ko, 2025-02-23:
> In my case it looks like SD card size is not the problem.

Yes, looks like random failure in other place, if i have more experience in STM32 debug it may help.

For now i have only theory:

On v2.9 settings stored inside STM32 in local EEPROM.
It have "very fast" reading and on bootloader stage read ALL settings take 10-100 cycles.
SDcard is not critical for bootloader so it can "Boot on default" and say "Bad SD card" etc.
So all "flags" as "Haptic ON" or "Teme to power on delay" available on moment of init CPU.

On v2.10+ this settings moved to SDcard (???). local EEPROM left untouched (i found "lost in update" models after downgrade to v2.9, so it says for me "EEPROM not used in 2.10".
I understand why it make like this? it remove limits of EEPROM and make possibility to add functional to SD and call "Big files" from SD, even settings for TX.

Why T12 is problem? 
Other users of EdgeTX use more powerful CPU on new TX with "Color screen".
I think in "Color" models CPU read SD card as fast as EEPROM and all needed data present in RAM on moment "bush PWR event"

On "B&W" models CPU is 90% STM32 and it pretty good for this usage (i not mining bitcoins on TX, i just want mix channels and send), but it may have "slower read time" for SD card.
So on weaker CPU on moment   "bush PWR event"  CPU have very small time (before it read EEPROM) to read all settings to RAM.

So my thinking about it (i am not professional programmer):

On v2.9 "Read settings time" almost immediate.
On v2.10+ "Read settings time" = CPU speed+SDcard read speed+load data to RAM speed

MUCH more operation to CPU.

If CPU "Cant read all data in time" bootloader try read  from memory flag "Haptic ON" but it return garbage,  and 50/50 or less chance to CPU fast enough to put all needed for "Complete BootLoader start seq.".

In my opinion in bootloader code absent "FailSafe - DEFAULT" values what can be read if SDcard not initiated yet.
And if "No needed data in RAM" - crash, if CPU can read fast enough settings and put in ram BEFORE bootloader  is initiated it crash in random states (0-2-4 "bars") and totally random pattern not related to SIZE of card, BUT depends "If CPU put settings in RAM" or not.

If bootloader may have "FailSafe DEFAULT values" that used if flag "SDcard read DONE" and prevent crash (but it load default settings, here i can only think "wait to SDcard" after "Complete bootloader exit" and if %wait_time% > 0.5S load in default and trigger warning "SDCARD TO SLOW!".

Difference in 2 diffetent T12 can be explained "Win silicone lottery" and my chip have a bit more chance to load settings.

I can be totally wrong, but maybe someone may look at code and confirm or discard my opinion.

upya4ko, 2025-02-28:
UPD:   tested tx12-a220ea3   ```1010011001```  -  Not fixed.

gherman59, 2025-05-02:
Installed v2.11.0 on my T12 Pro and it appears the power on problem is solved.  Thank you for your efforts!

upya4ko, 2025-05-09:
UPD: test v2.11.0 (**t12-8369a2e.bin**)
Bug exist: "111101101001100100100100110001"
15/15

later try build latest tag.

upya4ko, 2025-05-13:
UPD: I buy some small MicroSD cards and make more testing, bug still exist.

---

# Test results of old sdcards

```
sudo dd if=/dev/urandom of=/dev/sde bs=512K status=progress oflag=nocache oflag=sync
sudo dd of=/dev/null if=/dev/sde bs=512K status=progress oflag=nocache oflag=sync
sudo dd if=/dev/zero of=/dev/sde bs=512K status=progress oflag=nocache oflag=sync

sudo parted /dev/sde mktable msdos 
sudo parted /dev/sde unit b print free
sudo parted /dev/sde mkpart primary fat32 1024B 61997055B
sudo mkdosfs /dev/sde1 
sudo mount /dev/sde1 /mnt/usb/ 
sudo f3write /mnt/usb/
ramflush-disk 
sudo f3read /mnt/usb/ 
ramflush-disk 
sudo umount /mnt/usb/

sudo dd if=/dev/zero of=/dev/sde bs=512K status=progress oflag=nocache oflag=sync 
sudo parted /dev/sde mktable msdos 
sudo parted /dev/sde mkpart primary fat32 1024B 61997055B 
sudo mkdosfs /dev/sde1
sudo mount -t vfat -o rw,umask=000 /dev/sde1 /mnt/usb/ 
cp -rv ~/edgeFreshSD/* /mnt/usb/
ls /mnt/usb 
df -h
ramflush-disk
sudo umount /mnt/usb/

```

|Num| Vendor   | Cap   | DD Write | DD Read  | F3 Write  | F3 Read   | Start | End Bytes  | Pwr ON | Pwr FAIL | 
| - | -------- | ----- | -------- | -------- | --------- | --------- | ----- | ---------- | ------ | -------- |
| 1 | Nokia    | 64MB  | 1.9 MB/s | 3 MB/s   | 1.75 MB/s | 2.81 MB/s | 1024B | 61997055B  |    8   |    11    |
| 2 | Nokia    | 64MB  | 2.5 MB/s | 2.6 MB/s | 2.25 MB/s | 8.34 MB/s | 1024B | 61997055B  |    19  |    18    |
| 3 | Nokia    | 128MB | 10.5 MB/s| 12.1MB/s | 9.56 MB/s | 11.42 MB/s| 1024B | 127139839B |    3   |    10    |
| 4 | NoName   | 128MB | 4.0 MB/s | 6.6 MB/s | 3.84 MB/s | 6.19 MB/s | 1024B | 127139839B |    4   |    10    |
| 5 | NoName   | 128MB | 4.1 MB/s | 6.7 MB/s | 3.87 MB/s | 6.19 MB/s | 1024B | 127139839B |    8   |    10    |
| 6 | Nokia    | 256MB | 6.1 MB/s | 13.6 MB/s| 5.91 MB/s | 12.88 MB/s| 1024B | 255852543B |    3   |    11    |
| 7 | Kingston | 512MB | 5.4 MB/s | 6.9 MB/s | 4.74 MB/s | 10.56 MB/s| 1024B | 509083647B |    5   |    11    | 
| 8 | SanDisk  | 512MB | 5.2 MB/s | 5.3 MB/s | 4.77 MB/s | 11.05 MB/s| 1024B | 507379711B |    6   |    12    |
| 9 | Kingston | 512MB | 4.3 MB/s | 11.4 MB/s| 3.85 MB/s | 10.83 MB/s| 1024B | 512229375B |    7   |    10    |

philj404, 2025-05-19:
I have also seen this issue on my Jumper T12 Pro: _Jumper T12 Pro will not power on_
I have seen it on versions v2.11.1 and v2.8.4.

- Clearing the "Pwr ON/OFF Haptic" checkbox seems to make power-on more reliable.
- Also, the problem seems worse for me with a freshly-charged battery.

- on a possibly related issue, the default (uncalibrated) transmitter battery voltage was reading about 1V higher than I measured with a voltmeter.  That said, adjusting the battery voltage to read correctly did not make a difference.

upya4ko, 2025-05-21:
> and choose the "Flash Bootloader"
Hi, i use only this method.

> * Also, the problem seems worse for me with a freshly-charged battery.
Not sure, i using fresh 18650, and test with 100% charge and 30% (after many flashing and testing), in my case i not found correlations.

MrPepon, 2025-12-11:
somethin change?