EdgeTX/edgetx #4878: Edge TX buddy Froze while erasing firmware

HeavyJohnny, 2024-04-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

While updating the firmware on my TX16s Mk2, Edge TX Buddy Froze at 0% erasing Firmware. Now my TX is bricked. It will not power on and I can not get it into bootloader mode. When I attempt to power on the radio the orange lights scroll across the 1-6 lights about three times before it stops at the one position and nothing happens. When I plug in the USB cable I do see STM32 Bootloader in the device manager. Please help me undo this.

### Expected Behavior

I expected my firmware to be upgraded

### Steps To Reproduce

see error

### Version

2.7.1

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

11 pro

HeavyJohnny, 2024-04-14:
I was able to get it to work after loading 2.8 and then going forward to the latest firmware

pfeerick, 2024-04-14:
Since you could see the STM32 bootloader, the radio was in DFU mode. Were
you not able to then flash the radio with EdgeTX Buddy? As if you could
flash it with srm32cubeprogrammer, Buddy *should* have been able to flash
it then also.

The version of edgetx running at the time has no bearing on DFU mode - DFU
is built-in functionality of the STM32 MCU.

closed 2024-08-04 by github-actions[bot]