EdgeTX/edgetx #4879: [jumper t-lite v2] edgeTX v2.9.4 - stick drift

nightflasher, 2024-04-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

While flying the aircraft (miniKwad) keeps rotating on roll to the right, needs constant roll left input.
Calibration was done, ADC on or off doesn't matter, betaflight dead band is set to highest value ever - still happening. 

### Expected Behavior

No rotation over roll axis when no input is giving. 

### Steps To Reproduce

Load firmware to jumper t-lite v2 and fly

### Version

2.9.4

### Transmitter

Jumper T-Lite

### Operating System (OS)

Linux

### Anything else?

Downgrade to edgeTX v2.9.2 stopped the behavior.

3djc, 2024-04-16:
Please show us channel monitor values when this is happening

pfeerick, 2024-04-16:
Does it drift, or is it constant? Is trim zeroed?

nightflasher, 2024-04-20:
the drift is contant and there is no trim setting in betaflight or the transmitter
https://vimeo.com/936960448

3djc, 2024-04-20:
Betaflight is not where you should be looking for debugging , please show the values changing in channel monitor as asked earlier

nightflasher, 2024-04-21:
I've checked every possible corner, and betaflight is one too.

3djc, 2024-04-21:
Please show us values drifting in channel monitor, as requested several time now

nightflasher, 2024-04-21:
This is both straight after upgrade and stick calibration 
v2.9.4
[image]

v2.9.2
[image]

pfeerick, 2024-04-25:
Can you check both your inputs and mixes pages... you may find there is a "ghost" value that was restored in 2.9.4 that is causing it. 

Or just check against a new blank model... if when you check the channel monitor / mixes screen it doesn't have that extra roll input/offset, you know it is something wrong in the model settings for the ExternELRS model.

closed 2026-10-13 by github-actions[bot]