EdgeTX/edgetx #4880: [jumper t-lite v2] edgeTX v2.9.4 - wrong battery warning

nightflasher, 2024-04-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I've set up some logical switches for SD Log, TPower on change and vbat status every x seconds.

But on v2.9.4 it keeps yelling at me "battery critical" or "battery low" for no reason. It only happens after arming and bound to the aircraft.

### Expected Behavior

Just no yelling about battery when every single involved battery is at full state or at least 4v.

### Steps To Reproduce

Setup:
- delta change with 1mW for ExpressLRS and arming switch
- arming switch and play value vbat

### Version

2.9.4

### Transmitter

Jumper T-Lite

pfeerick, 2024-04-16:
As JC said, EdgeTx on it's own only cares about the transmitter battery voltage, which should not be affected by arm state (unless, in an extreme case like ELRS dynamic power option set, and it jumps from idle to max output power when you have a low battery). I would be checking against a new blank model, checking global and special functions for any play value lines that may cause it, and ensuring iNav telemetry Lua isn't running as it has it's own callouts also.

nightflasher, 2024-04-20:
>  third parties lua that could potentially issue those messages too
i have telemetry screen set up and i had inav lua running, which i thought was yelling at me, but i removed it and it didn't stop yelling at me. i have only set up some Logical switches, which should tell me the actual ExpressLRS transmit power and every X seconds the battery voltage, but it keeps yelling "battery low" and the vibration is shaking a lot.

after downgrade to 2.9.2 everything was workin fine the day. 

should i've mentioned, tha i've changed the gimbals to the ag01 mini? is it a point?

3djc, 2024-04-20:
You have been asked to check on a blank model without any lua or special function, what are the results of those tests?

nightflasher, 2024-04-21:
This is on a fresh and clean model.

3djc, 2024-04-21:
The only warning on clean model that exist are about transmitter batteries, not craft

nightflasher, 2024-04-21:
That's the thing I was wondering, because transmitter battery was on 4v all day.

pfeerick, 2024-04-25:
If you're still stuck on this, zip up your RADIO and MODEL folders and upload there here... with the model having issues as the active one. That may also help with the stick offset issue you have also.

nightflasher, 2024-04-29:
sorry for being late
[edgeTX_tlite_v2.zip](https://github.com/EdgeTX/edgetx/files/15147762/edgeTX_tlite_v2.zip)