EdgeTX/edgetx #4898: companion v2.10-rc3 loses switch settings from v2.9.2 file

arkarkark, 2024-04-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

when I load a .etx file from v2.9.2 companion into v2.10-rc3 some of my switches above the screen are lost.
see my first model (bee) Special function SF10 and SF11 - the Switch is missing in 2.10-RC3 and it's listed as TMR which is the first button/switch above the screen on my t-pro
both are configured  for the t-pro (v1)

here's the .etx file that I renamed to .zip so I can attach it.
[t-pro.zip](https://github.com/EdgeTX/edgetx/files/15045046/t-pro.zip)

### Expected Behavior

loding the v2.9.2 file into v2.10-rc3 should preserve all my settings.

it was [this thread](https://discord.com/channels/839849772864503828/1051165136473313290/threads/1230955556311797760) on discord that led me here.

### Steps To Reproduce

load attached file into v2.10 rc 3 companion
see how switches don't work.

### Version

Other (Please specify below)

### Transmitter

Jumper T-Pro

### Operating System (OS)

macOS

### OS Version

Sonoma

### Anything else?

this might be in the core edgeTX and not just companion since I just plugged my v2.9.2 jumper t-pro v1 sd card into my jumpter t-pro v2 running 2.10-rc3 and the buttons stopped working there too with no companion involved either.

elecpower, 2024-04-19:
Switches SE and SF are not included in the radio switch settings you provided and default to None for the radio, Mote: They do default to 2 Positions for T-Pro v2.

Steps to fix:
1. In Companion open the etx file
2. Select Edit Radio Settings
3. Select Hardware tab
4. Change switches SE and SF to 2 Positions
5. Close edit
6. Open the model and check the SFs.

Since Companion and radio firmware share the same default radio configuration this is likely why radio firmware is mirroring the same 'missing' switch behaviour (I do not have a T-Pro to test).

elecpower, 2024-04-19:
> Switch is missing in 2.10-RC3 and it's listed as TMR which is the first button/switch above the screen on my t-pro

For me it is displaying **Tmr1**. The label you have configured for the 1st button/switch is **TML**

arkarkark, 2024-04-19:
here it is in v2.9.2

[image]

here it is in v2.10

[image]

arkarkark, 2024-04-19:
@elecpower I started finding this problem when trying to move from  v1 -> v2, but the steps I list both companions are configured for t-pro v1 (ignore the profile NAME, sorry)

arkarkark, 2024-04-19:
@elecpower I misspoke in my first report, edited now. I'm pressing the TML button to tell me the battery telemetry and TMR1 timer

elecpower, 2024-04-20:
Thanks for the screenshots

elecpower, 2024-04-20:
I rechecked your uploaded zip and it contains this raw config which clearly has switch SE down for both SFs: (note: the item numbering is zero based)
[image]

Displays this after I set SE to 2 Positions
[image]

After I edited SF10
[image]

and it generates this
[image]

All as expected. Also TIMER1 -> Tmr1 plus the enable flag is an expected change in v2.10

So your screenshot from v2.9.2 does not appear to be based on the same file you uploaded.

arkarkark, 2024-04-20:
I made a video. I assure you it is the same file (I even included md5 sums for you)

it was too large to attach so I had to upload it here https://imgur.com/a/ENFGINZ

elecpower, 2024-04-20:
I can replicate your video using v2.9.4.
After more testing I can confirm the issue is not v2.10 but v2.9.x.
v2.9.x is writing and reading bad yaml. SW1 thru SW6 are being written as SE thru SJ and SE thru SJ is being read and mapped to SW1 thru SW6. So from the GUI all looks okay.

The major ADC rewrite for 2.10 has flushed out the bug.

This raises the question, how is the 2.9.2 firmware handling reading SE thru SJ? Also does v2.10 firmware handle SW1 thru SW6 correctly?

3djc, 2024-04-20:
@elecpower internaly, the function switches are designated as real switches, then mapped to function switches (https://github.com/EdgeTX/edgetx/blob/f0a50bd98f36cc72d14d8e461e6afb239d4e4271/radio/src/targets/taranis/hal.h#L738)

elecpower, 2024-04-20:
@3djc that might be ok for the firmware but Companion is reading the tpro.json file which does not include the mapping of the function switch eg SW1 to the real switch v2.9 SE or v2.10 SG and with the addition of the 2 extra option switches it makes things even worse when parsing the yaml.

```
    {
      "name": "SF",
      "type": "2POS",
      "flags": 0,
      "gpio": "GPIOG",
      "pin": "LL_GPIO_PIN_9",
      "default": "NONE",
      "display": [
        1,
        2
      ]
    },
    {
      "name": "SW1",
      "type": "FSWITCH",
      "flags": 0,
      "gpio": "GPIOG",
      "pin": "LL_GPIO_PIN_7"
    },
```

3djc, 2024-04-20:
Well you can compute it for 2.10. In this case, SWF is the last before SW1, so SW1 is also SWG. Unsure how the mapping is done for earlier versions

elecpower, 2024-04-20:
It needs to be in the json file so there is no need for Companion to have per radio rules which are prone to get out of sync with the firmware. This was one of the objectives of using shared json files.

> Unsure how the mapping is done for earlier versions. Is there some mapping table between versions?

I'm going to have to:
- add a specific conversion rule just for tpro for the 2 physical switch shift
- fix yaml parse and write
- fix gui switch lists from SW1:xxx to SG:xxx
- rework the loading of the json file???
- go down the rabbit hole...

3djc, 2024-04-21:
We can include in the json file the name of the switch for 2.10, albeit I have a feeling this is useless in this case, since 2.10 don't use those.

The issue remains for changes to existing radios over time. That's not the first time something like this happens. There has to be specific code written for it. Not too sure if trying to put this in a json file would be the answer, it would have to be manually generated anyway

elecpower, 2024-04-21:
The PR negates the need for inclusion in 2.10.
Eventually all json files will have to be generated manually as more and more entries get added as compiler directives are transitioned.

3djc, 2024-04-21:
Just commented the PR, I think there has been a misunderstanding there

closed 2024-04-25 by pfeerick

arkarkark, 2024-04-25:
thanks so much for all your work on this, it's much appreciated.

PR #4902 (2024-04-25): fix(cpn): T-Pro function switches
Fixes #4898

Note: I have not tested on hardware or simulator