EdgeTX/edgetx #4899: MT12 won't update to Centurion.

inventor7777, 2024-04-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

**I want to update my radio to the latest 2.10.0 RC3.** I loaded the mt12-30e8769.bin file onto the FIRMWARE folder. After I enter the boot loader and press the file, the MT12 says **"Not a valid firmware!"**

### Expected Behavior

I expected it to update to RC3.

### Steps To Reproduce

Add the .bin file to the sd card and try to install on MT12.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Operating System (OS)

macOS

### Anything else?

The version currently installed on my MT12 is the stock firmware, which was **2.10.0 Nightly**, according to the "About" screen.

pfeerick, 2024-04-19:
> Not a valid firmware!

You have tried re-downloading the firmware to your computer, and copying to the sd card again, and ensured safely removed/ejected/unmounted the sd card?  As this typically indicates the file copied to the card is corrupt.

inventor7777, 2024-04-20:
> > Not a valid firmware!
> 
> You have tried re-downloading the firmware to your computer, and copying to the sd card again, and ensured safely removed/ejected/unmounted the sd card? As this typically indicates the file copied to the card is corrupt.

I tried redownloading and replacing manually the old mt12 bin file. Still throws an error.

inventor7777, 2024-04-20:
I also tried updating to the RC2 and that didn't work either.

3djc, 2024-04-20:
Use buddy, because it does work fine here

inventor7777, 2024-04-22:
@3djc @pfeerick 

I tried using EdgeTX Buddy and it didn't work.

I opened the bootloader on the radio, plugged it into my Mac, (the radio said "USB Connected) added the .bin file to Buddy, went through the menus, and....it failed. It said that there was no USB DFU Interface.

https://github.com/EdgeTX/edgetx/assets/142270525/bc852f72-2eee-48e6-9899-8a615c72c9ec

pfeerick, 2024-04-22:
> (the radio said "USB Connected)

That's because you aren't supposed to have the radio turned on. Leave the radio turned off, plug in the USB, and see how you go then.

inventor7777, 2024-04-23:
@pfeerick Thank you very much! That solved the problem...I can't believe I couldn't figure that out 🙄

closed 2024-04-23 by inventor7777