EdgeTX/edgetx #4908: Invert Switches in Hardware menu

OckelboB16, 2024-04-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Possibility to invert the 2POS and 3POS switches.

### Describe the solution you'd like

In the SYS->HARDWARE menu: Inputs - Switches, there will be a new selection of Inverting the switch.

pfeerick, 2024-04-22:
Now that in 2.10 it will be possible to invert pots, there could be scope for this also for switches. 

What is a use case for doing this at the radio level, rather than just using inverted state of an switch?

OckelboB16, 2024-04-23:
The case at the radio level is for example how the switch input is displayed (x-value) on the curves graph.

3djc, 2024-04-23:
Not too fan of it. If you mod hardware, simply put the switch, or wire it, in the right position (as a reminder, the rule is "up is the closest to your head"

Sea-Kerman, 2024-05-13:
Would also mean you don’t have to boot the radio to check the orientation then desolder and resolder if you replaced a switch and put it in upside-down