EdgeTX/edgetx #4911: Moving FlightModes up/down

OckelboB16, 2024-04-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The possibility to move defined FlightModes up/down in the FLIGHT MODES menu,
and by doing that change the priority for the moved FM.  

### Describe the solution you'd like

When selecting a FM there will be a 'Move' choice.
Like the 'Move' choice in the MIXES menu.

pfeerick, 2024-04-22:
Order doesn't change priority - moving them would be an entirely cosmetic change. The "priority" is merely one of whether the enabling "switch" condition is true.  It would be nice though to be able to re-order them visually.

OckelboB16, 2024-04-23:
Try this
FM2 = SC down
FM3 = SC down AND SD down
compared to
FM2 = SC down AND SD down
FM3 = SC down
regarding priority.

3djc, 2024-04-23:
> Order doesn't change priority - moving them would be an entirely cosmetic change. The "priority" is merely one of whether the enabling "switch" condition is true. It would be nice though to be able to re-order them visually.

Sorry my friend, this is inaccurate

Quoting OpenTX manual: "The priority of the flight modes is such as the first FM of 1-8 that has its switch ON is the active one. When none has its switch ON, the default FM0 is active."

pfeerick, 2024-04-24:
True... (if you are daft enough to effectively make it so you only have at most two flight modes) but you _are_ quoting the wrong manual :stuck_out_tongue_closed_eyes:

pfeerick, 2024-04-24:
> FM2 = SC down
> FM3 = SC down AND SD down
> compared to
> FM2 = SC down AND SD down
> FM3 = SC down

How? Via an AND logical switch for SCdn and SDdn? As you can only use one switch for flight mode. If you do that, I see your point there... where first configuration only lets FM2 work regardless of SD position, whereas the second lets both FM2 and FM3 work. This is where I mispoke before, because the first valid match effectively blocks all the other FMs...

OckelboB16, 2024-04-24:
I just wanted to show the logic regarding FM priority.

The way you do it with EdgeTX is by using Logical Switches (Lnn).
L01: Function = AND ; V1 = SC down
L02: Function = AND ; V1 = SC down ; V2 = SD down
--
FM2: Switch = L01
FM3: Switch = L02

(For glider competition we have a lot of FM:s)

3djc, 2024-04-24:
> True... (although doesn't this mean you've been daft enough to also effectively make it so you only have two flight modes?) but you _are_ quoting the wrong manual, just in case you haven't noticed! 😝

I purposely did to show it has been like that for ages :)