EdgeTX/edgetx #4914: Stick positions displayed on the curve

OckelboB16, 2024-04-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The "stick" position displayed on the curve is the latest source choosen in the Mixes menu.
It would be a usable feature being able to also select the source when in the curve menu.

### Describe the solution you'd like

When, for example, pressing on the curve
there will be a list of the available sources that can be selected as
the input for the position displayed on the curve.

pfeerick, 2024-04-22:
When you edit a curve via the curves screen, it is not associated with any controls, hence why it automagically uses the last control to be moved. If you instead edit the curve via the inputs or mixes screens, since it is now linked to a specific source, it will track only *that* control.

MicheleVilla, 2024-04-27:
> When you edit a curve via the curves screen, it is not associated with any controls, hence why it automagically uses the last control to be moved. If you instead edit the curve via the inputs or mixes screens, since it is now linked to a specific source, it will track only _that_ control.

This option work with INPUT and MIXER page but not for OUTPUT Page, correct?
If I remember correctly there has been a lot of discussion about these options in the past

3djc, 2024-04-27:
There is no curve in output page, is there ?

pfeerick, 2024-04-27:
> There is no curve in output page, is there ? (nor should there be)

Yes, outputs have a curve also.

OckelboB16, 2024-04-27:
Off Topic: Curves on outputs (individual servos) is a very useful function.