EdgeTX/edgetx #4915: Sensor values when telemetry lost

OckelboB16, 2024-04-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When in the Special Functions menu having a sensor value,
for example the RQly, being played (sound) when a switch is asserted:
The old (latest) value is played when the telemetry is lost.
The sensor value is "red" in the Telemetry menu, meaning no updated values
due to lost telemetry.
 

### Describe the solution you'd like

Is it possible to instead of an old value,
have the value being 'none' due to lost telemetry.
i.e. the value played is replaced with the phrase 'none'.

pfeerick, 2024-04-22:
Or instead check if telemetry is active... that is another sensor value ;)

OckelboB16, 2024-04-23:
Maybe like this:
RQly = latest known value
RQly# = actual value (equals 'none' or 0 when telemetry lost)

pfeerick, 2024-04-25:
Do you have multiple telemetry items played? This would be the perfect candidate for a custom Lua script instead, in order to have the specific logic you want, rather adding yet another telemetry value (RQly, RQly-, RQly+, RQly#). 

Or, a logical switch that checks for your dedicated trigger switch and "Tele" to see if telemetry is active. You could have a second one that check for the switch and !Tele and have that play a phrase/tone if there is no value. 

Don't get me wrong, I'm not against adding stuff like this, just making sure alternatives have been thought through, to see if there is a better solution already present.

OckelboB16, 2024-04-26:
I was not aware of the 'Tele'. By using the 'Tele' my feature request can easy be solved with the current EdgeTX version.
Thanks!