EdgeTX/edgetx #4916: Trainer settings per modell and not system wide

Papa-Echo, 2024-04-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am the instructor in our club and I have several students for quadcopter (FPV) flying and for FPV planes. The problem is that OpenTx and Edge-TX only have one setting for Instructor/Student in the SYS menues.

### Describe the solution you'd like

I want to ask for a model specific Instructor/Student page.
The reason is that I teach several students on the same day. One student is a beginner and I limit his % to 50. Then I have a more advanced student who can fly with 100%. The next student comes with his own plane or quadcopter and has a different order of channels. Or the calibration needs to be changed.
Every time I need to change this SYS page.
It would be much better if I could set up a model for each student and set specific values for the Instructor/Student mode.

### Describe alternatives you've considered

This model trainer page could be added to the first model page if Instructor is selected.

pfeerick, 2024-04-25:
Sounds like another candidate for the global/model level settings paradigm being introduced.

sande005, 2024-05-12:
I noted in a past that having a few system trainer configs, that could be assigned to any model would be beneficial - ie "Bobs Radio, "Sue's Radio" that then could be selected (perhaps even by a switch) for a particular model.  OP carries this a step further and makes it all model specific.  So a complete model for Sue and another for Bob.  Can be limiting if the Instructor has several different models he will be using as a Master with the same students.  (I solved the rate problem by putting "low" on a dial, and then I can adjust the actual weight in the air to what suits that student).
The second example begs whether the students own plane/radio will be the master, or if the Instructor will setup and bind a new model so he can be the master.  If the first case, then the setup for the slave TX has to be done in the student radio - which this proposal doesn't solve.  OR beyond the work of duplicating/binding a new model in the Instructor radio, then either the OP's proposal or my variant will assist.
Mine, of course, seems to follow the design parameters of keeping Sources distinct from use of those sources.  Kind of like when Inputs was separated from the Mixer way, way back in the early days.

sande005, 2024-05-16:
@raphaelcoeffic - I agree with you, a one-stop config is generally preferable.  All of you are doing amazing work.  It will be interesting to see what solutions you can come up with for this classic one-to-many and many-to-one problem!  Whatever you do, rip up and throw away the current code in Companion for Trainers - it is very messed up (but was long,long ago in OpenTX also)

ChrisOhara57, 2024-09-10:
I've seen This request. I do not know whether someone is working on this.
I do a lot of training of new pilots in our club for 3 years now and I had the same problems.
Up to five students, different models, different radios.
I solved that problem by doing the calibration of the student/spotter radios in mixers instead of using the basic radio function.
The calibration in the basic function is always the same (offset=0, weight=100)

Originally I used copies of the models for different calibrations. But I ended up having a lot copies of the same models that made it difficult to keep them all updated to the same level.

Now I use only one model per plane. I do use scripts if I want to change from trainer to spotter version or from one students radio to another. (Some scripts are even automatically written by another script)

I know that this method would not be the same if you do some functionality in the basic code.
But probably you could find some ideas there that could be worth thinking about.

If you are interested I would upload a description (but this would take some time, cause I've to update it to this years modifications).

ChrisOhara57, 2024-09-17:
This is a description of my workarounds.

[Trainer-Student-Shortversion-English 2024-09-11.pdf](https://github.com/user-attachments/files/17025919/Trainer-Student-Shortversion-English.2024-09-11.pdf)

Papa-Echo, 2024-09-19:
Hi Chris,
Thank you for your input. I apologize for my late reply, but I am super
busy.

Although you found a workaround for the problem I still hope that the
developers take up my idea and change the EdgeTX code.

Thank you,
Peter

ChrisOhara57, 2024-09-20:
Hi Papa-Echo
I do agree. While my workarounds do work perfect I found that for the average (or new user) they are still to complex to understand.
We are ten OpenTX/EdgeTX user in our club and I am the only one who manage these things effortless.

Please do also have a look at my related proposal in issue #5532
Thanks you
Chris

ChrisOhara57, 2024-09-20:
> A very elegant, if complex solution. The one fly in the ointment - it is essentially an improved "quick take back" method. When the master moves a stick, he gets control. For those of us that "shadow fly" and keep the master sticks moving to the position they need to be in for the students current flight attitude, when master control is regained, this won't be useable. ie - student overbanks left, I move my stick right to anticipate a counter bank, but may not take control right away, as I give the student a few beats to correct on his own. It also seems to lend itself to lack of clarity about who is currently in control - say the master is at neutral, and the student hasn't done anything yet, student may be unclear that he again has control. I even have a voice prompt that announces "your plane" when I flip the switch to give the student control. But reality is I hardly ever enable it - the reassurance in my actually telling the student seems to help keep them focused.

I do agree. There are more than one way of how to gain the control for the trainer. Everyone has to find out what suites him best. 
While the discussion on this is very interesting I would suggest to focus in the  headline of this issue "Trainer settings per model and not system wide".

ror1948, 2025-01-29:
I agree that it would be better to have the SYSTEM Trainer screen within each model. I have a TAER slave and a AETR master and it is a pain to make it work without changing the SYS Trainer settings.
Do not think they ever should have been at the SYS level because we have so many different radio setups able to be SLAVE and MASTER.

ror1948, 2025-01-29:
> I don't want to have to configure the trainer stuff on each model. I configure the student radios with a trainer model, only used for trainer mode, so that the channels come in with the correct mapping.
> My kids have their own radios, so I have 3 different student radios connecting to my radios.

You would always have the option to use the global setting instead. Not everyone has the luxury of having dedicated training gear. Having one screen for all trainer settings would be a treat. Took me an hour to figure out why trainer was not working and I had forgot to set trainer master/jack option under MDL Trainer screen

thanks

ChrisOhara57, 2025-01-29:
I'm not sure that I understand what you are saying.

I do a lot of training with different models (different student and trainer radios, also some old 35MHZ as spotter or as student).

What I would prefer is that I do have to manage only one model for every plane.
Independent form the model I would like to store data of the different student radios (mapping and calibrating) in extra tables.

So when I start I would only have to select which Model I want to fly and  I have to select which one of the defined student radios I will use.

Is that your intention?

ror1948, 2025-01-29:
I have decided to use present system and just change the mapping in the global setting as required.
 Sent from AltaMail

ChrisOhara57, 2025-01-30:
So no possibility to save different Student radio settings and select them afterwards???

ror1948, 2025-01-30:
I can live with how it works now.

thanks