EdgeTX/edgetx #4926: Jumper T18 can not boot after flash 2.10 rc(1, 2, 3) via edgetx buddy

Curious-r, 2024-04-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Jumper T18 can not boot after flash 2.10 rc(1, 2, 3) via edgetx buddy. But if I flash it by bootloader method, it works well. Maybe this has something to do with the changes in the bootloader in 2.10.

### Expected Behavior

Boot normally

### Steps To Reproduce

Flash 2.10rc3 to T18 via edgetx buddy and then boot it.

### Version

Other (Please specify below)

### Transmitter

Jumper T18

### Operating System (OS)

Windows

### OS Version

Windows 11 23H2

3djc, 2024-04-26:
Just tested it, working fine here

Curious-r, 2024-05-01:
@3djc Hello!
Is your radio flashed via edgetx buddy?
In my radio, after press and hold the power button, the led will flash a while and then just stop.

3djc, 2024-05-01:
Yes, flashed with buddy

https://github.com/EdgeTX/edgetx/assets/5167938/1c3f9343-f697-4e59-b9f7-e62eaf669b5e

[image]

3djc, 2024-05-01:
Sorry but this radio is not supported, cannot help since we don't have any information or samples of this radio

closed 2024-05-01 by pfeerick

Curious-r, 2024-05-01:
[image]
Is this debug port? Maybe I can try with my jlink

pfeerick, 2024-05-06:
Unfortunately it seems that is only one of the two necessary IOs for SWD. It seems the pad labelled CLK is PA14 (SWCLK), but the one marked IO is PI3, whereas you want PA13 for SWDIO. Just waiting to hear back if that is available somewhere else.

pfeerick, 2024-05-06:
Actually, it seems like IO is actually SWDIO/PA13, so that should be usable as a SWD header.

Curious-r, 2024-05-06:
Thanks for help.
I have connected the SWD port to jlink, but there is nothing can be read via jlink CDC UART port currently. I'm checking what went wrong. I'll let you know if I get more information.

rotorman, 2024-05-06:
You are on the wrong path with UART. Let a debug build run e.g. via Segger Ozone with your J-Link and magic will happen.