EdgeTX/edgetx #4937: Enabling both internal and external ELRS should be usable or impossible

wimalopaan, 2024-04-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Oviously ETX is not designed to use two ELRS modules anabled at the same time. Telemetry -> "Start search sensors" completely hangs the radio (need to disconnect the battery). 

### Expected Behavior

It is possible to physically have two ELRS modules in one radio: internal and external.

Either:

a) If one of the modules is enabled, keep the other disabled

b) allow that both are enabled, and then use them in the expected behaviour. This includes telemetry (EdgeTx then has to process two telemetry streams) and the elrsV3.lua, which needs to choose between the two.

### Steps To Reproduce

1) Enable both modules for a specific model
2) In "Telemetry" start "Search sensors"
3) the radio freezes.

### Version

Other (Please specify below)

### Transmitter

FrSky X12

### Operating System (OS)

Linux

### OS Version

Arch Linux

pfeerick, 2024-04-29:
Didn't know it would make it crash, but I'm unexpected/indeterminate behaviour was expected, but is probably only documented deep in some telemetry related issues where telemetry needs to be overhauled to allow for seperate streams for multiple modules. Disabling both is not really an option even know, as you can use both (i.e. for dual independent Rx , or dual frequency), just can't manage telemetry or Lua with both connected at the same time. Similar to using multiple rxs with a single tx for extra channels, maybe disabling telemetry on second tx/rx pipelinemay help here.

wimalopaan, 2024-04-29:
Updating the internal and the external ELRS module to 3.3.4-RCx fixed the freeze problem.

raphaelcoeffic, 2024-04-30:
It shouldn't be missing much, as most things are already separate.

wimalopaan, 2024-04-30:
Well, I don't know, what the problems with the actual implementation is, so I can't imagine to nearby solution. I do not even know for sure if this is also true for the MPM.