EdgeTX/edgetx #4940: Sorting model images with roller causes emergency mode in TX16S with 2.9.4.

Luznatural, 2024-04-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have a Radiomaster TX16S with A01Hall, firmware 2.9.4 English language with a 4 in 1 internal multi-module, A01 gimbals, and Bluetooth mod.
Edgetx is custom made due to the need of bluetooth software. The firmware was dowloaded from the official github repository. (I didn't build it) 
When you try sorting model images using the roller to find the image you want for your model, it causes a lock of the screen, followed by a blanck screen and the message EMERGENCY MODE, appears 100% of the times in my Radiomaster TX16S.
If you try sorting with your finger, the issue does not occur.

### Expected Behavior

I expect to be able to quickly sort the images with the roller to localize faster which one I want for my model, without the radio going into emergency mode.

### Steps To Reproduce

1. Select: Radio settings
2. Go to File Manager and IMAGES folder
3. move the roller to the right (you can only go to the right) to sort the pictures (no need to make it too quickly).
4. After some images, the screen locks, then goes blank, and the message EMERGENCY MODE appears after a strong haptic and a beep.
5. You can turn off the radio, and the problem resets until you try again.
6. It happens 100% of the times.

### Version

2.9.4

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

philmoz, 2024-04-29:
Please zip up your images folder and upload here. You may have a bad image that is crashing the radio.

Luznatural, 2024-04-30:
Here you have it.
Most of the images were downloaded from Skyracoon
[IMAGES.zip](https://github.com/EdgeTX/edgetx/files/15157948/IMAGES.zip)
Thanks in advance

philmoz, 2024-04-30:
The images don't crash on my TX16S.
Does yours crash on specific images or randomly?
If you have another SD card might also be worth trying that.

Luznatural, 2024-04-30:
The crash occurs ramdomly. I will check with another card and let you know.
Thanks

Luznatural, 2024-04-30:
No luck! Changing card does not solve it. I also fresh formated both cards with fat32, just in case.
Apparently, the problem is related to the use of the scroll wheel. If I select the images touching the screen, the radio does not crash, but this system is way time consuming.

philmoz, 2024-04-30:
Very strange.
Do you have complex models with widgets and scripts? 
Can you try with a blank model selected.

Luznatural, 2024-04-30:
Tried with a blank model. No changes. The problem continues the same.

Luznatural, 2024-04-30:
I deleted the spaces and made the names shorter. Also, I deleted a single .jpeg file that I discovered in the list.
The problem continues, but it took a little longer to appear.
But for your information, I have exactly the same images and names (prior of modificating names and deleting anything) in my Horus X12S, and in that radio there is no problem to scroll through the images, other that having to wait a bit for the images appearing on the screen.
I guess the problem is especific of TX16S.

philmoz, 2024-05-01:
Is it only in the SD Browser that it crashes, or does it also crash in the file chooser popup (i.e. when you want to select the model image in the model setup page)?

Luznatural, 2024-05-01:
It only crashes when the system shows pictures in the screen. When you select an image for your model, the screen does not show the picture, and it doesn't crash. I will make a more detailed survey throughout different menus to check and I will tell you.

philmoz, 2024-05-01:
If you time it might be worth testing 2.10 RC3 (make sure you have a full backup).

Luznatural, 2024-05-01:
I have tested with 2.10 RC3 and the problem apparently is solved in that version. But it comes back when returning to 2.9.4
I hope release version of ETX 2.10 would be available soon....

pfeerick, 2024-07-06:
Since it appears to be fixed in 2.10, closing.

closed 2024-07-06 by pfeerick