EdgeTX/edgetx #4950: CPN: flex switches don't seem to be supported at all

3djc, 2024-05-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Flex switches don't seem to be handled companion side

### Expected Behavior

Flex switch support

### Steps To Reproduce

TX16S with 2,10 RC3
Go to hardware menu
Assign in Pots EXT1 and EXT2 to switch
In Switches, assign FL1 and FL2 to respectively EXT1 and EXT2, define a type (2 or 3 pos)
Create a special function triggered by either FL1 or FL2

With companion, read radio and models, go to check the special functions: empty

### Version

Other (Please specify below)

### Transmitter

RadioMaster TX12 / TX12MK2

### Operating System (OS)

macOS

### OS Version

Sonoma

elecpower, 2024-05-01:
I thought I did have this working in Companion but seems I need to recheck. Could be reading models and settings that is broken.

3djc, 2024-05-01:
To save you time, this is my test model

[model2.yml.zip](https://github.com/EdgeTX/edgetx/files/15176240/model2.yml.zip)

elecpower, 2024-05-01:
@3djc I also need your radio.yml please

elecpower, 2024-05-02:
Companion has the hardware configuration so on to yaml....

3djc, 2024-05-02:
Here you go. Done with  2.10RC3 on a TX16S

[MODELS+RADIO.zip](https://github.com/EdgeTX/edgetx/files/15184634/MODELS%2BRADIO.zip)

elecpower, 2024-05-02:
Loaded your latest zip and it all looks okay to me. What am I missing?

[image]

[image]

[image]

3djc, 2024-05-02:
omg, what an unlikely scenario ! Here is what happened:

- before trying this I was working on companion support for yet unannounced radio (in corresponding branch). Ra dio type was set to this radio
- i switched to 2.10 branch to test the above
- build tx16s libsimu, companion, launched it
- was greated by (to be expected)
[image]
- then downloaded model and settings, and look two things: SF are missing, and title bar says TX16S
[image]
- but I did find out that I also have this
[image]

After selecting the TX16s as radio model, it does indeed work
[image]

3djc, 2024-05-02:
So conclusion reported work did not exist, a different (much smaller, and only affecting developers) issue exists

closed 2024-05-02 by 3djc

pfeerick, 2024-05-02:
> So conclusion, reported issue did not exist, a different (much smaller, and only affecting developers) issue exists

Not just developers... Neil has mentioned in the past that if a radio doesn't exist, for historical reasons Companion falls back to assuming TX16S... we're just unlucky enough to encounter it more often (and not read the instructions in bold).

3djc, 2024-05-03:
I did, but since the title bar said TX16S, which is what I wanted anyway, I assumed that it defaulted to TX16S.  Ideally, it should be either blank, or show what it defaulted to. But a minor thing I agree

elecpower, 2024-05-03:
I'll see what I can do to improve.
However when you use an app in bleeding edge mode you have to expect things to go pear shaped from time to time 🤣

3djc, 2024-05-03:
> I'll see what I can do to improve. However when you use an app in bleeding edge mode you have to expect things to go pear shaped from time to time 🤣

Couldn't agree more, and clearly RTFIIB should also apply to me !!!!

pfeerick, 2024-05-03:
It is clearly trying to confuse and befuddle... and has succeeded on more than one occasion! :hand_over_mouth: 

Even if you add a flashing neon light... I guarantee you'll me complaining about this again also :zany_face: 

Maybe make it fall back to a definite UNKNOWN_BOARD type state, if not going to basically hard fail till the user provides sensible input? :innocent: