EdgeTX/edgetx #4956: Horus X10 E ACCESS radio does not regain control of receiver after going into EM (Recovery Mode)             ETX V2.10RC3

BompaBert, 2024-05-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

By scrolling to and selecting a model in the model manage screen, radio went into EM. Power off/on required to get out of the situation- chosen model was not selected.
See also issues / fixes #3603, 3599 and 4940.

### Expected Behavior

Upon triggering EM, radio should regain control of the receiver within  a few seconds.

### Steps To Reproduce

1. Open manage model screen - long enter [MDL].
2. Scroll through models and select one by pushing [ENTER}

### Version

Other (Please specify below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

pfeerick, 2024-05-04:
If I make you some test firmware for 2.9.4 and 2.10.0-RC4, that will cause a EM on demand, are you in a position where you can test if your receiver(s) will come back up in emergency mode? i.e. do you have a setup that can safely have no signal from handset when on the ground? It could be a simple as grabbing another receiver, BEC and servos, and making a small test platform to see if you get controls back, and setup a new model on the transmitter just for that.

BompaBert, 2024-05-05:
Yes, am glad to be of help. Sent from my iPhone

pfeerick, 2024-05-05:
Okey dokey... when you come onto github to read this message, you'll find attached a zipfile with a 2.9 and 2.10 firmware for the X10 Express

With these firmwares, there is an additional special function you can access, bottom of the list of functions, simply called "Test". Once that special function is enabled/active, the radio should crash pretty much straight away... i.e. I usually make the active condition my moment switch if I need to use it. I would get your RF link all sorted out, setup the special function, and then restart the radio (to ensure everything is saved, as any unwritten configuration settings may be lost). Then you won't have to re-add the special function if it gets clobbered. 

[x10-express-debug-firmwares.zip](https://github.com/EdgeTX/edgetx/files/15211789/x10-express-debug-firmwares.zip)

BompaBert, 2024-05-05:
@pfeerick, with both firmware's test function, the radio goes into 'emergency mode', throttle and all control surfaces drop and as expected control is regained after a few seconds. For the test, the same airplane and identical configuration was used as it was when control was lost. Good to know though, the 'sticky' function drops too but does not come back as it needs triggered again - a bit of a challenge with no visual control or sound. Also, with your firmware 'Emergency Mode' is displayed while it was 'Recovery Mode' previously.

3djc, 2024-05-05:
So all seems according to expectations

BompaBert, 2024-05-05:
Yes, just that it puzzles me why am seeing  'Emergency mode' instead of 'recovery mode'. Has this been changed in the debug firmware's?

3djc, 2024-05-05:
The 'recovery' word is not part of EdgeTx at all, so if you saw it, it was coming from a lua of some kind

BompaBert, 2024-05-05:
All lua's on the radio are from our website downloads page, no customs. It baffles me that a lua could put the radio, besides not regaining control and different wording,  into exactly the same EM situation as experienced in today's test -black screen and the need of a power off/on to regain control.

pfeerick, 2024-05-05:
Thanks for testing that... so it looks like Emergency mode is working correctly, and bringing back the link and mixer - no UI or Lua. I can't remember if logical switches or special functions will funciton... but I would operate on the basis that they won't - only simple mixer controls. Even if they did, they would be in the same state they were when the radio turns on, not how they were before the reset. 

Ok, lets re-wind a for minute... so you are saying you had a black screen with white text saying "Recovery Mode" instead of "EMERGENCY MODE" when you were having issues before? Specifics matter here. Since as JC said, "recovery" does not appear in any form in the firmware that would be shown on screen - not even for translated firmware. And if a Lua shows it, you're not in Emergency Mode, as Lua does not function **at all**.

BompaBert, 2024-05-05:
Yes, confirmed, screen was showing 'recovery mode'. The logicals work in emergency mode as well. I use a button with sticky with an edge and a special function to enable and disable throttle. EM drops the sticky but I could re-enable it again in EM.

pfeerick, 2024-05-06:
Hm... have absolutely no idea where that is coming from then! 😖 There is no text "recovery mode" at all in the project source **at all*, and the nearest match is *recover*, but this is a completely difference screen (i.e. similar to the throttle/switch warning alerts), which also explains that radio settings are corrupt, etc, not simple black and white screen.

BompaBert, 2024-05-09:
Got the radio working on 2.10RC4. Will inform you once I get an EM again.

BompaBert, 2024-06-14:
Radio went into Emergency Mode again, version 2.10.1 with SD Card V2.10.0. Using a sticky with an Edge to enable/disable throttle got me in trouble. The EM drops the sticky and I lost throttle. Not knowing my radio was in EM I belly landed the bird.
So am removing any critical sticky's cos when flying you don't realize what's happening as the radio does not gives any signal when it goes into EM. I believe Ethos EM does. If you think there is a way to create an alarming distinctive sound for EM situations that would be great, it would reduce confusion and could have saved the bird damage. If feasable, I
[image]
[image]
 can open a suggestion, plse let me know. Attached EM pictures. Unable to attach SDcard here - too big.

philmoz, 2024-06-14:
2.11 will restore Sticky switch state in an EM.

BompaBert, 2024-06-14:
@Philmoz,  any chance to produce a sound or haptic at or just after the EM
condition?

philmoz, 2024-06-14:
I think this has been discussed before.
While it's possible, I would be concerned that an unexpected sound or haptic could be quite distracting to anyone who did know what it meant.

BompaBert, 2024-06-14:
Jee, Ethos does it though. IMO, there should be a warning. Just my
experience today, I had no clue I was in EM. When you fly, you look at your
bird, not the radio screen. If I would have known, my bird could have been
saved. Also, it does not feel right to fly in EM mode anyway.  To me, a
audible warning different from those we have already would be a +. Maybe a
param in the setup to enable/disable?

BompaBert, 2024-06-14:
Or you could make it a Special Function. Then the pilot can decide what to
do with it. A .wav that would do it for me "Warning, Emergence Mode"...

3djc, 2024-06-15:
I would simply not fly with a radio that does EM

BompaBert, 2024-06-15:
Taken in account everything that has been stated here, allow me to suggest following. And I do not mind opening a new suggestion if needed. Powering the radio on, we are greeted with "Welcome to EdgeTX". What about when a EM situation occurs, the radio informs the pilot using the same voice saying "Emergency Mode active" or something similar. No distraction, just purely informative...Just that flying in EM is something you do not want to do...

3djc, 2024-06-15:
No wav can be played since sd is not mounted in EM

BompaBert, 2024-06-15:
Gosh yea, thanks for the reminder. You got my vote though on a haptic or whatever is feasible to pronounce EM. The logic here is that an EM is always unexpected and the pilot should be brought aware somehow he is flying in EM. Now, only the display shows but when something goes wrong during flight and it's all hands on deck, the last thing on your mind is to look at the display. That said, we good as you can circumvent the sticky drop for now. Everything else works as it should during EM. Thanks for 'sticking' with me here :-)