EdgeTX/edgetx #4957: Display of curves during processing - Transmitter Radiomaster TX16S

Gubi70, 2024-05-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When changing the Y/XS input, the + / - menu logically appears. However, this means that the lower points of the curve are no longer displayed.
This makes setting the "lower" points very time-consuming, as the curve graphic cannot be moved up/down.

### Describe the solution you'd like

The option to move the curve graphic up or down so that the points to be edited can still be saved

philmoz, 2024-05-02:
If you are using a radio with a roller (rotary dial) then you can change the values by pressing the roller (ENTER key) then using the roller to adjust the value. This does not open the popup number editor.

Gubi70, 2024-05-03:
tested and works. Thanks for the quick help/info.