EdgeTX/edgetx #4961: The center position of the S1 button on the screen does not correspond to its actual center

pierrotm777, 2024-05-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The center position of the S1 button on the screen does not correspond to its actual center.
Even if I use the control setting menu, the center is shifted on the screen.

I use 2.10 RC3 version !!!

### Expected Behavior

Correct this difference between position of the S1 button and that on the screen

### Steps To Reproduce

Always

### Version

2.9.4

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

[image]

pfeerick, 2024-05-03:
So you did Radio Settings -> Hardware -> Calibration, and you centred the pots, sliders and sticks, and then moved them full range, as prompted?

rotorman, 2024-05-03:
Also check if Pot with detent is checked, or just Pot, as with just Pot, only 2 end positions are used for calibration and with Pot with detent, 3 positions - ends and center.

pfeerick, 2024-05-03:
Radio Settings => Hardware => Pots

pierrotm777, 2024-05-03:
I have tried to use Calibration with the same result, the index isn't centred.

pierrotm777, 2024-05-04:
Sorry for this delay.
These are my models.
[MODELS.zip](https://github.com/EdgeTX/edgetx/files/15209182/MODELS.zip)

3djc, 2024-05-04:
and RADIO ?

pierrotm777, 2024-05-04:
TX16S MKII

3djc, 2024-05-04:
RADIO folder please

pierrotm777, 2024-05-04:
[RADIO.zip](https://github.com/EdgeTX/edgetx/files/15209215/RADIO.zip)
I had found that 3 models was in 2.11.0 but i use the 2.10.0 RC3 firmware.
model31.yml
model23.yml
model21.yml

3djc, 2024-05-04:
Just to be sure, on what model are you seen it ?

3djc, 2024-05-04:
I think you are not calibrating correctly.
I loaded your fiels and I had the issue
Redid calibration, worked fine

You need to ensure your center pots and slider when displaying "Center stick...."
and move them all to both extremes when displaying "Move axis..."

pierrotm777, 2024-05-04:
I use S1 for change the VOLUME.
So i had defined a Global Function
[image]

3djc, 2024-05-04:
Please read my message above, you are likely not calibrating correctly

pierrotm777, 2024-05-04:
Ok, my tx16s battery is empty :-(.
I try that as soon as possible

pierrotm777, 2024-05-04:
I had used the calibration incorrectly.
It's now ok :-(

Thanks for your help !!!

Pierre

closed 2024-05-04 by 3djc