EdgeTX/edgetx #4964: Add MT12 joycon button press to usable inputs

Camo55, 2024-05-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

My hope is to be able to use the joystick module to its full potential

### Describe the solution you'd like

Right now I cannot map a switch to the spare joystick module's momentary switch on the MT12

### Describe alternatives you've considered

Currently the radio only supports S3 and S4 for both swappable modules, but the joystick module has an additional third function that the dual-switch module doesnt: momentary press

3djc, 2024-05-04:
Unfortunately, there hardware radio side doesn't support it, it cannot be done

pfeerick, 2024-05-04:
As JC said, the hardware itself doesn't support that - so it is not something we can implement. The button is not electrically connected to the handset at all.

closed 2024-05-04 by pfeerick