EdgeTX/edgetx #4965: Frsky X-lite Pro modifying menu items swiping right

Camo55, 2024-05-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

https://github.com/EdgeTX/edgetx/assets/17286162/22de1f25-1dcb-4cdf-a9f8-79f4c3871096

Sliding to the right while picking a letter results in the next letter being 'automatically' picked

### Expected Behavior

Menu item is supposed to remain on the chosen setting when tapping sideways. Pressing to close the selection works normally.

### Steps To Reproduce

Open menu,
Make menu text selection
Tap joystick to the right

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X-Lite / S / Pro

### Operating System (OS)

Windows

### Anything else?

Nightly build 5-4-2024, 2.10 RC4

pfeerick, 2024-05-04:
This also affects BetaFPV LR3 - it stems from https://github.com/EdgeTX/edgetx/pull/4884 and radio with LEFT and RIGHT keys

philmoz, 2024-05-04:
T-Lite is also affected. Can you test PR #4967 on X-Lite radio.

pfeerick, 2024-05-04:
Sorry, got caught up with another bug so thought they weren't affected. T12 and LR3Pro were affected by this also, and this resolves it for the LR3Pro on hardware, and X-Lite S on simulator, so just waiting for confirmation from OP. Also resolves my remaining bugbear from #4884 regarding popup menu navigation quirk.

Camo55, 2024-05-05:
> Sorry, got caught up with another bug so thought they weren't affected. T12 and LR3Pro were affected by this also, and this resolves it for the LR3Pro on hardware, and X-Lite S on simulator, so just [...]

Is there a .bin file I can download to try the firmware update? I am a noob to github.

pfeerick, 2024-05-05:
https://github.com/EdgeTX/edgetx/suites/23447572357/artifacts/1473568468

For future reference: 

> To access the automatic builds for a Pull Request (PR) you are interesting in trying out, first open up page for the PR in question. Under the title for the pull request, you should see a series of tabs (Comments, Commits, Checks, Files Changed). Click on the Checks tab. 
> 
> If you are after the radio firmware, click on the link on the left side mentioning Firmware. 
> Or, if you want a Companion/Simulator build, pick the link on the left side mentioning Companion that suits your operating system. 
> 
> Then, under the 'Artifacts' heading near the bottom of the page will be a link to the zip file containing either the firmware or companion package.

Camo55, 2024-05-05:
Thank you! I have installed the firmware and can confirm the issue has been resolved!

PR #4967 (2024-05-05): fix(bw): editing names on 9x/xlite nav radios & popup menu using wrong keys/events
Fixes #4965
closed 2024-05-05 by Camo55