EdgeTX/edgetx #4969: cpn: mt12 support

bd4, 2024-05-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No companion support for mt12 yet

### Describe the solution you'd like

Can use companion with mt12

### Additional context

I didn't see any issue or PRs related to MT12 companion support, thought it might be useful to have an issue to track progress and coordinate contributions?

Wondering if there is some way I can contribute, I understand there are a lot of differences so may be a fair amount of work.

3djc, 2024-05-05:
Yes, a lot of work is involved, since companion was designed a (long) while ago with only air radio in mind

pfeerick, 2024-05-05:
If we take any shortcuts, it will not be pretty. For sure, we could just slap in support for the MT12 like any other radio, but then people would be complain about the constant references to air stuff in companion, and drive stuff on the radio, and then there is simulator :laughing:

jimdabim, 2024-06-03:
Hi there,

any way you could make the MT12 work together with the companion would be very much appreciated.
Even if it is not perfect in a first version and the simulator would look like an air one.

It would still halp a lot.

Greetings,
Sebatian

sirwatchalot, 2024-06-04:
I agree. Much of the learning curve for EdgeTX (especially for surface guys new to it) is in navigating the menu's and working out what things do. Companion greatly simplifies this and allows for very rapid "tweek then simulate" learning. Having Companion available would hugely help adoption in my view. If it isn't already in the plan, the I fear the MT12 using EdgeTX was perhaps a mistake...

magnusroe, 2024-06-04:
> If it isn't already in the plan, the I fear the MT12 using EdgeTX was perhaps a mistake...

It has said in the roadmap that support is in the plan since at least January. I agree it would be nice to get it as soon as possible, but maybe semi-functional features would throw off new adopters just as soon as they were attracted. If I'm not mistaken a lot of the surface segment typically use one tx/rx pair per vehicle. Maybe the functionality can be introduced in betas though? The current page is a little confusing when it mentions support will come with 2.10 and we're already at 2.10.1

sirwatchalot, 2024-06-04:
I agree - beta releases would be good. It would also help to know what the current timeline is. I've no issue with the Companion updates lagging a bit, but the early comments above seem to suggest work hasn't even started. Hopefully that's not the case.
(and being simplistic, it's the GUI that needs to change, the EdgeTX behaviours don't)

bd4, 2024-06-04:
My understanding is, it is in the plan but is a lot of work because of differences with air radios. I wish I had more time to contribute but have pretty inconsistent free time to devote to it, and am not familiar with the code yet. I am sure more developer resources would be welcome.

I do think that a limited release with no simulator but basic editing functionality would still be useful to people.

sirwatchalot, 2024-06-04:
I concur - even the just the editing functionality would be useful and lower the bar to entry.
But as for the radios being different - sure the combination of pots, switches, etc. is different but the radio doesn't actually DO anything different for each of those (except where we config EdgeTX differently). So it should just be a visualisation problem, no?

gagarinlg, 2024-06-04:
It is a bit more. Some texts have to be redone, depending on the radio type.
@elecpower did a lot in the last couple of months to make it easier to introduce new radios into companion, but there is still a long way to go, to make it as easy as it should be t add new radios.

3djc, 2024-06-04:
Also, quite a few things expect 2 sticks (4axis) when MT12 only has 2

sirwatchalot, 2024-06-04:
> It is a bit more. Some texts have to be redone, depending on the radio type. @elecpower did a lot in the last couple of months to make it easier to introduce new radios into companion, but there is still a long way to go, to make it as easy as it should be t add new radios.

Ah - so it's as much about it being difficult to add a new radio (regardless of type) as it is about it being surface vs air

On the 2 axis sticks vs single axis - from a simulator perspective what breaks if nothing ever drives the sticks? Or is something hard baked? From an abstraction perspective I'd assume there are some archetypes that get assembled (pots, switches, 2 axis sticks) that have visual behaviours. Beyond the visual, has the existence of 2 sticks been baked as an assumption into something unfortunate?

sirwatchalot, 2024-07-05:
is there any actual work being done to provide full support for the MT12?

sirwatchalot, 2024-08-04:
Any update? I see 2.10.3 has been released again without Companion support and I still don't understand what makes simulator support so hard other than the visuals. The wheel and trigger are directly analogous to sticks after all, and underneath those, EdgeTX is still EdgeTX.

pfeerick, 2024-08-04:
Companion support is more likely to be added to 2.11 than 2.10.x.

You, or anyone else, are welcome to implement  support if it is "so easy".
In addition to the Sim UI later visuals and input controls, all throughout
Companion there is assumption that there are 4 primary controls, which was
perfectly valid since this has been the case since day one for Companion.,
and no UI concept of surface (for all the terminology changes).

sirwatchalot, 2024-08-05:
You quote "so easy" which hasn't ever been said. I sense some bitterness for some reason.

But your response does suggest there isn't actually any work in progress and, lacking any other info, also suggests there isn't likely to be. That doesn't bode well for the MT12 or any other novel usage of EdgeTX.  And as above, I still don't understand what impact there is from an assumption of 4 controls when each of those controls is independent (with the mixer and outputs creating dependencies). Is there any architectural documentation that explains?

pfeerick, 2024-08-05:
> You quote "so easy" which hasn't ever been said. I sense some bitterness for some reason.

No, not bitterness... perhaps frustration over the "I still don't understand what makes simulator support so hard" comment when when no less then three developers have already said it is not a simple task. :rofl: And no, I don't think there is any documentation other than the Companion code itself.

sirwatchalot, 2024-08-05:
Let's just say that professionally I find that people saying things are hard without explanation 50% of the time means they don't want to do it rather than it actually being hard. Not saying that is the case here, but it does tend to colour perceptions. As per earlier in the thread the intention was to know that it was even being worked on, and no one was able to say it was, which also wasn't a good sign.

3djc, 2024-08-05:
I always find it also quite amazing that some people feels they are entitled to have some right to decide what unpaid volunteers should be working on ....

bd4, 2024-08-05:
> Let's just say that professionally I find that people saying things are hard without explanation 50% of the time means they don't want to do it rather than it actually being hard.
I understand you're frustration in lack of companion support. Please be respectful of all the time that unpaid developers put in to this so that everyone can benefit. AFAIK radiomaster developed the hardware and did not pay anyone to develop the software, so it is down to volunteers deciding to invest their time.

sirwatchalot, 2024-08-05:
> I always find it also quite amazing that some people feels they are entitled to have some right to decide what unpaid volunteers should be working on ....

that's the nature of "community" isn't it? Allowing people to have a say? And where above to you read "a right to decide"? It isn't there, and for good reason. 

Though the conversations become very much easier if current timelines are clear, and for the MT12 they weren't (having "yet" in the release notes isn't hugely informative. ). All this thread started as was a question about when. In the lack of "when" it became about "whether", that's all.

Though, and to bd4's point, one might have imagined that Radiomaster would have at least have ensured full support for the radio...

PR #5381 (2024-09-26): feat(cpn): add support for surface radios inc Radiomaster MT12
Fixes #4969

Summary of changes:
- add support for surface radios
- add support for Radiomaster MT12
- move lcd capabilities from firmware to boards
- minor fixes found during testing
- more post ADC refactor fixes
- housekeeping

TODO
- [x] surface radio configuration
- [x] Companion yaml encode/decode own models and settings
- [x] surface radio decode Companion encoded yaml
- [x] Companion decode radio encoded yaml
- [x] surface radio simulator
- [x] MT12 default settings eg battery, screen
- [x] MT12 radio configuration
- [x] MT [... cut]
closed 2024-09-26 by pfeerick