EdgeTX/edgetx #4972: 6-pos calibration v2.10rc1-4

jrwieland, 2024-05-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

During calibration of the 6 position switch remains at the highest setting "6" pressed and does not cycle through the selections 1-5 when pressed again.

### Expected Behavior

During calibration the selection matches the button pressed after cycling through them and can be verified before leaving calibration.

### Steps To Reproduce

Hardware >>Pots>>
6POS>>choose (Multipos Swicth) >>enable it

Start Calibration
Cyle through the 6 position switch 1,2,3,4,5,6
Press a lower number to verify the calibration 
number stays at 6 (or the highest number pressed during testing)

If you choose slider (as the 6 position type) it cycles through the slider bar, back/forth

### Version

Other (Please specify below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

Firmware tested on 
tx16s-0f7b0ec-210rc1
tx16s-58b8cb3-210rc2
tx16s-30e8769-210rc3
tx16s-e0aca64-210rc4
and Current nightly

3djc, 2024-05-06:
Works ok here, could you backup your RADIO folder, then delete it and retry?

jrwieland, 2024-05-06:
@3djc 
Strange Behavior - when I delete the radio folder and create new it works.
when I upgrade to 2.10 it doesn't calibrate.  That's why I went through each release. so narrow when it happened.
When I downgrade back to 2.9 everything calibrated normally.

However when I delete the radio folder as suggested and create new it works (v210rc4).

Closing the issue since it may only be on my end doing several upgrades/downgrades to narrow it and ended up corrupting the radio files. (this is why we have backups)

closed 2024-05-06 by jrwieland