EdgeTX/edgetx #4974: Switches on Taranis QX7 not visible to select in "Special Functions"

dadiowe, 2024-05-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Switch SH showing in Radio settings but not in Model settings, Setup Page, Warnings section.

Pots S1 and S2 showing in Radio settings and Model settings, Setup page but not a selectable option in the Special Functions Switch Column dropdown.

In Radio Settings, Switches there are 2 non existent switches, SI & SJ

### Expected Behavior

The switches, pots should be in the listed places

### Steps To Reproduce

Open Edge TX companion and select relevant pages.

### Version

Other (Please specify below)

### Transmitter

FrSky Q X7 / Q X7S (ACCST)

### Operating System (OS)

Windows

### OS Version

10

### Anything else?

This is a completely new install of edge Tx companion 2.10 and a completely new install of matching firmware, Sound files and SD card files to a newly formatted 1GB SD card.
I believe the problem was present before the upgrade from companion V2.9.1 but I can't be sure.
Thanks for all your hard work on this great software.

pfeerick, 2024-05-06:
Except, pots are not switches, so that part is an invalid request. Only
switches will be shown in the switches field!!

I'll have to check the rest on radio.

3djc, 2024-05-06:
Couple of comments before bed time

SI and SJ are place holder for P400 and P401 header on X7 PCB

SH is a toggle, so should not appear in warning section

pfeerick, 2024-05-06:
Thanks JC, saves me checking the hal:) So it also has support for two extra switches, nice to know! And if they are configured, they do show, so that part is working fine. 

Toggle (momentary) switches do not show in the switch warnings section, hence why SH doesn't show.

closed 2024-05-06 by pfeerick

dadiowe, 2024-05-06:
Thanks for the response.
I see what you mean. Just trying to figure out how to use a pot for volume or brightness control.

philmoz, 2024-05-06:
[image]

dadiowe, 2024-05-06:
Thanks, it is always easy when you are shown.
Kindest regards
Rob