EdgeTX/edgetx #4975: Switch indicator location on b/w displays

Camo55, 2024-05-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Misaligned arrows

### Describe the solution you'd like

[image]

### Describe alternatives you've considered

Leaving it alone / using maximum 2 letters for switch name

philmoz, 2024-05-07:
The arrows are always displayed immediately to the right of the name so this is not going to change.

PR #4976 adjusts the switch label position so it will fit long names.

Camo55, 2024-05-07:
Center-justified text will work

PR #4976 (2024-05-07): fix(bw): adjust 128x64 layout to accommodate long switch names
Fixes #4975 

Summary of changes:
- Adjust position of switch names to allow more room
- Center justify switch names so it looks more balanced.

[image]
[image]
closed 2024-05-07 by pfeerick