EdgeTX/edgetx #4977: Colorlcd - add `us` as a PPM Unit option

pfeerick, 2024-05-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In https://github.com/EdgeTX/edgetx/issues/3163, it was noted that `-DPPM_UNIT_US` did not work on colorlcd, i.e. changing display and inputs from `%` to `us`, as well as asking if it could be made so that it was not something required a custom build. https://github.com/EdgeTX/edgetx/pull/3993 addressed the latter point, but the "PPM Units" runtime entry only gives options of `0.0` and `0.--` on colorlcd - which correctly reflects the units supported on the colorlcd UI at present. 

### Describe the solution you'd like

That a third option be added, which allows for units to be displayed as `us` where they are currently shown only as `%`, as well as for relevant edit fields to change from `%` to `us` input. i.e. this should be changed to match B&W level of support. 

### Describe alternatives you've considered

Digging out my old slide rule

sande005, 2024-05-07:
FWIW - all three option are available in Companion, and work correctly.  But after download to radio (Frsky X10S-E Access) all revert to the default.  As noted "us" option is not listed on the radio.

xros, 2024-05-24:
> ### Is there an existing issue for this feature request?
> * [x]  I have searched the existing issues
> 
> ### Is your feature request related to a problem?
> In #3163, it was noted that `-DPPM_UNIT_US` did not work on colorlcd, i.e. changing display and inputs from `%` to `us`, as well as asking if it could be made so that it was not something required a custom build. #3993 addressed the latter point, but the "PPM Units" runtime entry only gives options of `0.0` and `0.--` on colorlcd - which correctly reflects the units supported on the colorlcd UI at present.
> 
> ### Describe the solution you'd like
> That a third option be added, which allows for units to be displayed as `us` where they are currently shown only as `%`, as well as for relevant edit fields to change from `%` to `us` input. i.e. so it matches the B&W level of support.
> 
> ### Describe alternatives you've considered
> Digging out my old slide rule
> 
> ### Additional context
> _No response_

same issue on 2.9.4 (self-compiled with `-DPPM_UNIT_US`) too.

pfeerick, 2024-05-24:
`us` units was never fully implemented on colorlcd anyway as part of the colorlcd rewrite from opentx 2.4, but https://github.com/EdgeTX/edgetx/pull/4987 for colorlcd side should finally implement that.

PR #4987 (2024-06-02): feat(color): support PPM_US setting
Fixes #4977 and #3163

I've tried to match all the places that the PPM_US setting is used on B&W radios.

Channel monitor. Note if using PPM_US setting the value at the right above each channel changes to % instead of showing us value.
[image]

Custom failsafe settings:
[image]

Outputs list (min/max values):
[image]

Outputs widget:
[image]
closed 2024-06-02 by pfeerick